Yacubane/esp32-arduino-matter #1: Test with different boards

oleksiikutuzov, 2022-11-07:
Hey, thanks for this project! I could successfully build the firmware also with PlatformIO.

Updating to the latest CHIP seems to have fixed the interaction with HomeKit, now I may try to create some other examples (however, it's strange for now. E.g. switch device doesn't seem to be supported by HomeKit yet or I am doing something wrong).

I have ESP32-C3 and S2 boards and could test with them. I guess S2 isn't officially supported, but I wanted to let you know.

jakubdybczak, 2022-11-07:
Hey! I'm happy that somebody actually used this lib and is working. I wasn't able to test HomeKit because I don't have such hub. Google does not support Matter at 1.0 (it supports some old Matter branch which is not compatible), so I'm not able to test end to end this lib with these ecosystems right now. According to https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md **switch** should work.

You are right, S2 is not supported, my fault ;)
Regarding C3, I have created test [esp32c3-test](https://github.com/jakubdybczak/esp32-arduino-matter/tree/esp32c3-test) branch which does compile with Arduino IDE. Unfortunately I don't have ESP32-C3 to be able to test it. Most probably it won't work. In case of regular esp32 I had to manually select which precompiled libraries are needed to make it work.
If you want to test it, please remember setup cpp17 support for esp32c3.

I hope to get esp32-c3 in future, but firstly I would like to focus on support for esp32. It looks like Matter is far from being mature, even like you said upgrading to latest CHIP does helped... but the old version was also based on v1.0 branch which is supposed to be compliant with Matter v1.0 specification. It seems like there are still some bugs that cause problems and incompatibilities between versions based on same specification. I also noticed that old CHIPTool wasn't working with newer version of CHIP from the same v1.0 branch.

astorks, 2022-11-08:
Thanks for this library, it's so much easier than messing around with the idf.

I was able to test out the [esp32c3-test](https://github.com/jakubdybczak/esp32-arduino-matter/tree/esp32c3-test) branch and can confirm it is working perfectly on an [M5Stamp C3U](https://shop.m5stack.com/products/m5stamp-c3u-mate-with-pin-headers). I also have a few [TinyS3's](https://esp32s3.com/tinys3.html) and a [ProS3](https://esp32s3.com/pros3.html) that I can test out an esp32s3 if you're able to add support for that as well. Anyway here is all the apps I've tested so far.

### Android ChipTool
Successfully commissioned the device over "ble" and I'm able to read and write to the device as well.

### Android Google Home App v2.60.60.4
Successfully commissioned the device over "ble" and "onnetwork", however I am not able to control the device through the google home app (I just get a message saying "Something went wrong") but I think that has to do with the google home app not fully supporting matter 1.0 yet as I have the same issue with all matter devices I have tried so far even built directly from [esp-matter](https://github.com/espressif/esp-matter) with idf.

### Home Assistant 2022.11.1 - Matter Server v0.3.0
Does not support matter 1.0 yet. Unable to commission a device, device console reports "matter chip[SVR]: Failsafe timer expired".

jakubdybczak, 2022-11-08:
Thanks for your feedback!
I can also confirm same issues with Google Home -- commisioning works but after that I get "Something went wrong" error.
Differences with HomeKit might occur due to specific sdkconfig (https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/sdkconfig), against which (slightly modified) I compile `esp-matter`.

I will try to find some time at the end of week to prepare version for S3 and check what is happening with C3 on PlatformIO.
I also have some scripts that can generate this library but there is a big mess with these scripts that I need to clean before releasing ;)

oleksiikutuzov, 2022-11-09:
>  check what is happening with C3 on PlatformIO.

It seems to be an issue with linking esp_matter
```
...
Linking .pio/build/esp32c3/firmware.elf
/Users/oleksiikutuzov/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pio/build/esp32c3/src/Light.ino.cpp.o: in function `esp_matter::cluster::general_diagnostics::config::config()':
/Users/oleksiikutuzov/git/esp32-arduino-matter/examples/Light/.pio/libdeps/esp32c3/ESP32 Arduino Matter/src/esp_matter_cluster.h:117: undefined reference to `esp_matter::node::create(esp_matter::node::config*, int (*)(esp_matter::attribute::callback_type, unsigned short, unsigned long, unsigned long, esp_matter_attr_val_t*, void*), int (*)(esp_matter::identification::callback_type, unsigned short, unsigned char, void*))'
/Users/oleksiikutuzov/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pio/build/esp32c3/src/Light.ino.cpp.o: in function `esp_matter::cluster::occupancy_sensing::config::config()':
/Users/oleksiikutuzov/git/esp32-arduino-matter/examples/Light/.pio/libdeps/esp32c3/ESP32 Arduino Matter/src/esp_matter_cluster.h:380: undefined reference to `esp_matter::endpoint::occupancy_sensor::create(unsigned int*, esp_matter::endpoint::occupancy_sensor::config*, unsigned char, void*)'
/Users/oleksiikutuzov/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /Users/oleksiikutuzov/git/esp32-arduino-matter/examples/Light/.pio/libdeps/esp32c3/ESP32 Arduino Matter/src/esp_matter_cluster.h:380: undefined reference to `esp_matter::cluster::get(unsigned int*, unsigned long)'
/Users/oleksiikutuzov/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pio/build/esp32c3/src/Light.ino.cpp.o: in function `setup()':
/Users/oleksiikutuzov/git/esp32-arduino-matter/examples/Light/src/Light.ino:86: undefined reference to `esp_matter::attribute::get(unsigned int*, unsigned long)'
/Users/oleksiikutuzov/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /Users/oleksiikutuzov/git/esp32-arduino-matter/examples/Light/src/Light.ino:91: undefined reference to `esp_matter::endpoint::get_id(unsigned int*)'
/Users/oleksiikutuzov/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /Users/oleksiikutuzov/git/esp32-arduino-matter/examples/Light/src/Light.ino:94: undefined reference to `esp_matter::start(void (*)(chip::DeviceLayer::ChipDeviceEvent const*, int))'
/Users/oleksiikutuzov/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /Users/oleksiikutuzov/git/esp32-arduino-matter/examples/Light/src/Light.ino:97: undefined reference to `PrintOnboardingCodes(chip::BitFlags<chip::RendezvousInformationFlag, unsigned char>)'
/Users/oleksiikutuzov/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pio/build/esp32c3/src/Light.ino.cpp.o: in function `get_onoff_attribute_value()':
/Users/oleksiikutuzov/git/esp32-arduino-matter/examples/Light/src/Light.ino:102: undefined reference to `esp_matter_invalid(void*)'
/Users/oleksiikutuzov/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /Users/oleksiikutuzov/git/esp32-arduino-matter/examples/Light/src/Light.ino:107: undefined reference to `esp_matter::attribute::get_val(unsigned int*, esp_matter_attr_val_t*)'
/Users/oleksiikutuzov/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pio/build/esp32c3/src/Light.ino.cpp.o: in function `set_onoff_attribute_value(esp_matter_attr_val_t*)':
/Users/oleksiikutuzov/git/esp32-arduino-matter/examples/Light/src/Light.ino:113: undefined reference to `esp_matter::attribute::update(unsigned short, unsigned long, unsigned long, esp_matter_attr_val_t*)'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32c3/firmware.elf] Error 1
```

jakubdybczak, 2022-11-11:
I have created new [new_variants_and_cleanup](https://github.com/jakubdybczak/esp32-arduino-matter/tree/new_variants_and_cleanup) branch. Added support for ESP32S3 and cleaned up lib files (library will now be smaller). 

I have tested compilation process on PlatformIO with `board_build.mcu` set to `esp32c3` and `esp32s3` without problems. @oleksiikutuzov could you post your platform.ini configuration that has this issue?

oleksiikutuzov, 2022-11-11:
@jakubdybczak now it builds without any issues from this new branch

jakubdybczak, 2022-11-19:
I have got this email tonight, looks like we will be able to test Google Home in next few days ;)

> Hi Google Home Developer,
> 
> With the release of the [Matter 1.0](https://csa-iot.org/newsroom/matter-arrives/) standard by Connectivity Standards Alliance last month, we are preparing Google Home to support Matter soon.
> 
> Today, we are excited to share that we will be rolling out our release in the Developer Preview to support the Matter 1.0 standard. We will begin pushing out this release on November 21 and continue through December. Any [Nest Hub (2nd gen)](https://store.google.com/product/nest_hub_2nd_gen) you have opted in for the Developer preview will receive this update.
> 
> Once your [Nest Hub (2nd gen)](https://store.google.com/product/nest_hub_2nd_gen) receives the updated build, previously commissioned Matter devices need to be re-commissioned. They will also need to have a compatible firmware with the [Matter 1.0 specification](https://github.com/project-chip/connectedhomeip/tree/v1.0-branch) to function correctly. Otherwise, Pre-1.0 software will be unable to pair and work with Google Home.
> 
> If you have any additional questions, please visit [our support page](https://developers.home.google.com/matter/support). 
> 
> Thank you,
> Google Home Developer Relations

satishmunot, 2022-11-24:
Arduino: 1.8.15 (Windows 10), Board: "ESP32-WROOM-DA Module, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled"

Library esp32-arduino-matter-master has been declared precompiled:

Using precompiled library in C:\Users\abc\Documents\Arduino\libraries\esp32-arduino-matter-master\src\esp32

Sketch uses 1549141 bytes (118%) of program storage space. Maximum is 1310720 bytes.

Global variables use 92460 bytes (28%) of dynamic memory, leaving 235220 bytes for local variables. Maximum is 327680 bytes.

Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.

text section exceeds available space in board

Error compiling for board ESP32-WROOM-DA Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

jakubdybczak, 2022-11-24:
@satishmunot I can see that you have selected default partition scheme in which this library unfortunately won't fit. Please change it to larger one, according to [Readme](https://github.com/jakubdybczak/esp32-arduino-matter/blob/master/README.md), for example to `Minimal SPIFFS`.

ruddycanuck, 2022-11-30:
Just a note to say I used the main branch & Arduino to get this working with the ESP32-C3-DevKitM-1U dev board and Smartthings. It worked right away. Thanks!

AGRO666, 2022-12-15:
I tested it with HomeKit
I set esp_log_level_set("*", ESP_LOG_ERROR);
the pair was not created

E (61577) chip[DL]: Long dispatch time: 2792 ms, for event type 7
E (67245) chip[DL]: Long dispatch time: 598 ms, for event type 7
E (70826) chip[DL]: Long dispatch time: 1720 ms, for event type 7
E (72219) chip[DL]: Long dispatch time: 1163 ms, for event type 7
E (73544) chip[DL]: Long dispatch time: 1228 ms, for event type 7
E (73715) wifi:sta is connecting, return error
E (73717) chip[DL]: esp_wifi_connect() failed: ESP_ERR_WIFI_CONN
E (74357) wifi:sta is connecting, return error
E (74359) chip[DL]: esp_wifi_connect() failed: ESP_ERR_WIFI_CONN
E (74360) chip[DL]: Long dispatch time: 640 ms, for event type 3
E (76710) chip[DL]: Long dispatch time: 1713 ms, for event type 3
E (77160) chip[SC]: Received error (protocol code 2) during pairing process: 54
E (77162) chip[IN]: CASE Session establishment failed: 54
E (79066) chip[DL]: Long dispatch time: 1722 ms, for event type 3
E (79215) chip[SC]: Received error (protocol code 2) during pairing process: 54
E (79217) chip[IN]: CASE Session establishment failed: 54
E (127467) chip[SVR]: Failsafe timer expired
E (127470) chip[SVR]: Commissioning failed (attempt 1): 32
E (127553) chip[ZCL]: OpCreds: Got FailSafeTimerExpired
E (127555) chip[ZCL]: OpCreds: Proceeding to FailSafeCleanup on fail-safe expiry!
E (127556) chip[FP]: Reverting pending fabric data for fabric 0x1
E (127625) chip[FP]: Warning: metadata not found during delete of fabric 0x1
E (128340) chip[FP]: Warning: metadata not found during delete of fabric 0x1
E (128403) chip[ZCL]: OpCreds: failed to delete fabric at index 1: d8
E (128405) chip[DL]: Long dispatch time: 937 ms, for event type 32782
E (128408) chip[BLE]: HandleChipConnectionReceived failed, err = d
E (128412) chip[BLE]: failed handle new chip BLE connection, status = d
E (142547) chip[BLE]: no endpoint for unsub recvd

AGRO666, 2022-12-15:
I tested in Homekit  debug.ino
on_off_light - worked
window_covering_device - worked
fan - worked
thermostat - worked
generic_switch - Not supported
on_off_plugin_unit - worked (in log - E (1498) chip[ZCL]: Duplicate attribute override registration failed)
Other accessories I not tested

oleksiikutuzov, 2022-12-15:
I can also confirm that all accessories except the generic_switch from Debug.ino worked in Apple Home, however rgb light requires additional clusters (there's an example in esp-matter). The switch should also work somehow, as it already does for the Hue hub.

palsbo, 2023-01-12:
I made a ESP device with your lib at it works. I further developed it to support color-light. Is a bit confused about color-temperature ligt and color-light, but it is not a big issue right now. 
I tested on EVE for Homekit on IOS and managed to paire. 
I also tested with Google Home. It works partly, but problems are with Google I think. Google Home preview currently only works with Matter on Android - not IOS (and only some Andoid phones??).
Preview version of Google Home is not working with cam for Q-code reading, but manually entering the code makes it pair.
Now the problem. I use Arduino IDE. When compiling, I have to enable 'Erase all flash' to make it possible to pair. 
Google Home loses connection during night(!?), and then I have to recompile to pair again. Is there a way to put the device in pairing mode (by using a pin/button?

mamarguerat, 2023-01-12:
Hi there! Thanks for the library and the example code. I tried to use it with an ESP32-C3 on a Seeed Studio Xiao. I can't get it working. When compiling with PIO I get the error (after a lot of warning about redefined #define).
```
/src/include/lwip/opt.h:51,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/system/SystemConfig.h:176,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/ble/BleConfig.h:43,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/lib/core/CHIPConfig.h:42,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/lib/core/CHIPCore.h:28,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/app/MessageDef/InvokeRequestMessage.h:22,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/app/CommandSender.h:30,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/app/DeviceProxy.h:29,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/esp_matter_core.h:17,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/esp_matter_attribute.h:18,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/esp_matter.h:22,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/Matter.h:9,
                 from src/main.cpp:1:
C:/Users/<username>/.platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32c3/include/soc/include/soc/soc_memory_types.h: In function 'bool esp_ptr_byte_accessible(const void*)':
C:/Users/<username>/.platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32c3/include/soc/include/soc/soc_memory_types.h:75:17: error: 'SOC_EXTRAM_DATA_LOW' was not declared in this scope
     r |= (ip >= SOC_EXTRAM_DATA_LOW && ip < (SOC_EXTRAM_DATA_HIGH));
                 ^~~~~~~~~~~~~~~~~~~
C:/Users/<username>/.platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32c3/include/soc/include/soc/soc_memory_types.h:75:17: note: suggested alternative: 'SOC_RTC_DATA_LOW'
     r |= (ip >= SOC_EXTRAM_DATA_LOW && ip < (SOC_EXTRAM_DATA_HIGH));
                 ^~~~~~~~~~~~~~~~~~~
                 SOC_RTC_DATA_LOW
C:/Users/<username>/.platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32c3/include/soc/include/soc/soc_memory_types.h:75:46: error: 'SOC_EXTRAM_DATA_HIGH' was not declared in this scope
     r |= (ip >= SOC_EXTRAM_DATA_LOW && ip < (SOC_EXTRAM_DATA_HIGH));
                                              ^~~~~~~~~~~~~~~~~~~~
C:/Users/<username>/.platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32c3/include/soc/include/soc/soc_memory_types.h:75:46: note: suggested alternative: 'SOC_RTC_DATA_HIGH'
     r |= (ip >= SOC_EXTRAM_DATA_LOW && ip < (SOC_EXTRAM_DATA_HIGH));
                                              ^~~~~~~~~~~~~~~~~~~~
                                              SOC_RTC_DATA_HIGH
C:/Users/<username>/.platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32c3/include/soc/include/soc/soc_memory_types.h: In function 'bool esp_ptr_in_iram(const void*)':
C:/Users/<username>/.platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32c3/include/soc/include/soc/soc_memory_types.h:105:28: error: 'SOC_CACHE_APP_LOW' was not declared in this scope
     return ((intptr_t)p >= SOC_CACHE_APP_LOW && (intptr_t)p < SOC_IRAM_HIGH);
                            ^~~~~~~~~~~~~~~~~
C:/Users/<username>/.platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32c3/include/soc/include/soc/soc_memory_types.h:105:28: note: suggested alternative: 'SOC_RTC_DRAM_LOW'
     return ((intptr_t)p >= SOC_CACHE_APP_LOW && (intptr_t)p < SOC_IRAM_HIGH);
                            ^~~~~~~~~~~~~~~~~
                            SOC_RTC_DRAM_LOW
In file included from C:/Users/<username>/.platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32c3/include/esp_event/include/esp_event_legacy.h:21,
                 from C:/Users/<username>/.platformio/packages/framework-arduinoespressif32@3.20005.220925/tools/sdk/esp32c3/include/esp_event/include/esp_event.h:28,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/platform/ESP32/CHIPDevicePlatformEvent.h:30,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/platform/CHIPDeviceEvent.h:343,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/platform/ConnectivityManager.h:32,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/platform/ESP32/ConfigurationManagerImpl.h:28,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/platform/ConfigurationManager.h:219,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/platform/CHIPDeviceLayer.h:27,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/app/CASESessionManager.h:27,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/app/InteractionModelEngine.h:62,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/esp_matter_core.h:18,
                 from .pio/libdeps/seeed_xiao_esp32c3/ESP32 Arduino Matter/src/esp_matter_attribute.h:18,
[... cut ...]
I use the `platformio.ini` from [issue 6](https://github.com/jakubdybczak/esp32-arduino-matter/issues/6#issuecomment-1339482081) as I had the same issue, now resolved.
```
[env]
platform = espressif32@5.2.0
framework = arduino
lib_deps = https://github.com/jakubdybczak/esp32-arduino-matter.git
build_unflags = -std=gnu++11
build_flags = -std=gnu++17
board_build.partitions = min_spiffs.csv
monitor_speed = 115200
upload_speed = 921600

[env:seeed_xiao_esp32c3]
board = seeed_xiao_esp32c3
```
I hope getting it working with your help!

mamarguerat, 2023-01-12:
@jakubdybczak I found the error !
I didn't include the `Arduino.h` file in my code. That's because it isn't present in your example code. Thank's for your help!

palsbo, 2023-01-13:
Thanks – it works Sendt fra Mail til Windows Fra: Jakub DybczakSendt: 12. januar 2023 18:39Til: jakubdybczak/esp32-arduino-matterCc: Erik Palsbo; MentionEmne: Re: [jakubdybczak/esp32-arduino-matter] Testing discussion (Issue ***@***.*** We created discussion about resetting device and putting into pairing mode is here: #12

palsbo, 2023-01-14:
You have made a greate job with this library. Thank you!
I have a hard time finding my way around the code. I tried to find a way to test if the device has been paired, but with no luck. Can you help me?

I try to pair with several apps. Some works. Some works sometime and some does not work (most of the time).
Homekit is the most reliable. Eve for homekit is also ok. Google Home does not work (at the moment), but it changes all, the time.  It seems we at at a very early state of development. 

Thanks again for the good work.

jakubdybczak, 2023-01-14:
Happy to hear it @palsbo! 

I found some logic that disables BLE (and so pairing as well) when WiFi is already provisioned here: https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/nimble/BLEManagerImpl.cpp#L150 Maybe it will be enough for your case? Keep in mind that "WiFi being provisioned" does not necessary mean that accessory works correctly with Matter controller, but at least it gives an information if WiFi credentials were correctly sent in pairing process. Unfortunately, I could not find better solution for your problem. You can try it:

`DeviceLayer::ConnectivityMgr().IsWiFiStationProvisioned();`

Regarding Google Home I can see and experience various problems with it. Sometimes it works and sometimes it's not. I think the only thing that we can do right now is to send logs to Google or wait until it's more stable. Many people have problem with Google services (https://github.com/espressif/esp-matter/issues/145, https://github.com/espressif/esp-matter/issues/156)

AintLarry, 2023-02-13:
> I made a ESP device with your lib at it works. I further developed it to support color-light. Is a bit confused about color-temperature ligt and color-light, but it is not a big issue right now.
> I [...]

Can you post your SourceCode of the colorlight?

palsbo, 2023-02-13:
I use this to reset into factory-reset. 
`
DeviceLayer::ConfigurationMgr().InitiateFactoryReset();
`
Activatet by a button to a pin. 
Google Home and Matter is just not working yest. Yes you can pair it but there are so many problems.

AintLarry, 2023-02-14:
How do you Implement the ColorLight / LightTemperture?

palsbo, 2023-02-14:
I implemented my example on a ESP32 with a small color TFT and 2 buttons. [Here](https://www.tinytronics.nl/shop/en/development-boards/microcontroller-boards/with-wi-fi/lilygo-ttgo-t-display-v1.1-esp32-with-1.14-inch-tft-display) The code is a bit messy, but I will give you my four files. Hope you can get some ideas from it.:

File ColorLight.ino:

`#include <qrcode_espi.h>

#include <Matter.h>
#include <app/server/OnboardingCodesUtil.h>
#include "tft.h"
#include "color.h"
#include "but.h"

using namespace chip;
using namespace chip::app::Clusters;
using namespace esp_matter;
using namespace esp_matter::endpoint;

/*
  Configuration:
  Board:            use ESP32_WROOM_DA (even with physical device TTGO
  Core Debug Level: none
  Partition Scheme: Minimal SPIFFS(.....

  https://github.com/jakubdybczak/esp32-arduino-matter

  I (922) chip[SVR]: SetupQRCode: [MT:Y.K9042C00KA0648G00]
  I (927) chip[SVR]: Copy/paste the below URL in a browser to see the QR Code:
  I (933) chip[SVR]: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3AY.K9042C00KA0648G00
  I (943) chip[SVR]: Manual pairing code: 3497-011-2332

   https://www.rapidtables.com/convert/color/hsl-to-rgb.html
   https://www.programmingalgorithms.com/algorithm/hsv-to-rgb/cpp/

  All collor values are in 0..255 range. To use color.h, convert hue by hue*360/255, sat by sat/256, vol by vol/256
*/

//#define ESP_LOG_LOCAL ESP_LOG_INFO
#define PAIRED DeviceLayer::ConnectivityMgr().IsWiFiStationProvisioned()
#define PLACE1 10, 30, 220, 95, 10

static HSVTORGB hsv;

TFT tft;

BUT but0(35);
BUT but1(0);

int last_toggle_r;
int last_toggle;
bool powerState = false;
bool paired = true;

// Cluster and attribute ID used by Matter light device
const uint32_t ON_OFF_CLUSTER_ID = OnOff::Id;
const uint32_t ON_OFF_ATTRIBUTE_ID = OnOff::Attributes::OnOff::Id;

// Endpoint and attribute ref that will be assigned to Matter device
uint16_t color_light_endpoint_id = 0;
attribute_t *attribute_ref;

//
//  Identificateion
//
static void on_device_event(const ChipDeviceEvent *event, intptr_t arg) {
  if (paired != PAIRED) {
    paired = PAIRED;
    if (!paired) qrcode.create("MT:Y.K9042C00KA0648G00");
    else {
      tft.fillScreen(TFT_BLACK);
      tft.setTextColor(TFT_WHITE, TFT_BLACK);
[... 477 lines cut ...]
    void drawCentreString(char * buf, int y = 15) {
      display.drawCentreString(buf, cx, y, cs);
    }
    void drawCentreString(String buf, int y = 15) {
      display.drawCentreString(buf, cx, y, cs);
    }
    void centerText(String buf, int x = cx, int y = 15, int c = cs) {
      display.drawCentreString(buf, x, y, c);
    }
    void msg(const char * msg, int y = cy) {
      display.fillScreen(0xBDF7);
      display.setTextColor(TFT_BLACK, 0xBDF7);
      display.drawCentreString(msg, cx, y, cs);
    }
    void qCreate(String msg) {
      qrcode.create(msg);
    }
    void begin() {
      display.init();
      display.setRotation(3);
      qrcode.init();
    }
};

#endif`

AintLarry, 2023-02-20:
Thank you.

Here is what I made out of it.
[XESP32C3_MTTR_RGB.zip](https://github.com/jakubdybczak/esp32-arduino-matter/files/10787984/XESP32C3_MTTR_RGB.zip)

Works Fine with AppleHome. But I wasnt able to connect to GoogleHome.

palsbo, 2023-02-21:
Google Home does not yet pair on iPhone (IOS), but you can pair on Androit and then it will work on IOS too.

AintLarry, 2023-02-21:
I tried to pair it on android, nur GoogleHome drops a Message, that the device isnt Certified.

palsbo, 2023-02-21:
About Google Certification. This is a bit stupid about google. I think you need to register a test-device. Look at this: https://developers.home.google.com/matter/test#create
Go to the Developer Console
Create a Matter Test project
Add matter integration, I added two: Vendor ID 0xff1 for both. Product ID 0x8000 and 0x0001.
I don't have the deeper understanding why(?).

AintLarry, 2023-02-21:
Oh, thats a little mess, cause i build this device for a Good Friend. 
    i am using ios/ipadOs, but he is an android user.

so now i have to create a matter test project on his Account? Yeah…..

are There Any other possibilities?

AintLarry, 2023-02-21:
Short Update:
I created an account, added the test device a now I can connect my light to Google home. But I can’t control it…
Maybe it’s because I don’t own a compatible hub?

Cause I’m an Apple user I don’t want to buy one. I’m happy with my HomePod mini. :D 

do you know any third partie android app to control matter devices **without** any hub?

jakubdybczak, 2023-03-12:
>Add matter integration, I added two: Vendor ID 0xfff1 for both. Product ID 0x8000 and 0x0001.

Google changed something, because I didn't have to add it two months ago. Now it's impossible to add not-certified or not-test device to Google account. Adding product with ID 0x8000 was enough for me.

ba1yuu, 2023-04-23:
Hi, i am a student who wants to try to connect matter with esp32 to control leds, but I see that your sample program cannot be connected. I would like to ask if there is a backup.

shabunin, 2023-09-07:
tried to compile for m5stickc plus (esp32-pico), lib version v1.0.0-beta.3 (because m5 libraries uses esp 2.0.7).

light example compiled successfully, but 

```
Sketch uses 1579277 bytes (120%) of program storage space. Maximum is 1310720 bytes.
```

I won't try to reduce it (i guess it's better to purchase  board), just reporting.

palsbo, 2024-02-28:
> > Add matter integration, I added two: Vendor ID 0xfff1 for both. Product ID 0x8000 and 0x0001.
> 
> Google changed something, because I didn't have to add it two months ago. Now it's impossible to add not-certified or not-test device to Google account. Adding product with ID 0x8000 was enough for me.

So will it be a problem to create devices with other pairing-code/QR-code and use them with Google? 
If not - how could we implement more devices with different parirng-code/QR-code?

It looks like something has to be done with the following two lines: (from the examples)
`#include <credentials/examples/DeviceAttestationCredsExample.h>
  esp_matter::set_custom_dac_provider(chip::Credentials::Examples::GetExampleDACProvider());
`
but how?

MxFrs, 2024-02-29:
I am using the following code, which is mostly the code @AintLarry uploaded, to control my 'ambient light' with the Alexa App. Unfortunately the code is not responding to any white color requests. I can only chose colors, no neutral white or warm white other dedicated white values that are available in the app. Is that a known issue or am I missing something in my code?

Second issue which is very weird and maybe has something to do with my 2.4G Wifi is that yesterday I was able to connect using the Apple Home App instead of the Alexa App. Today, the Apple App is not working anymore, no matter which code I upload to the ESP32. If anybody has an idea to fix that, that would be amazing as well.
[ColorLight.zip](https://github.com/Yacubane/esp32-arduino-matter/files/14451254/ColorLight.zip)