Yacubane/esp32-arduino-matter #3: Switch Example

scottgarner, 2022-11-25:
Super grateful for the work you've done so far on this!

I have lights up and running with HomeKit based on your example, though the connection process feels pretty slow and unreliable. I'm trying to get a simple switch working with zero luck and I wondered if you might have some insight. 

I've tried something like this with both an on/off switch and a generic switch. At best the accessory will show up in HomeKit with the message "This accessory is not curerretly supported by the Home app".

```c++
  {
    generic_switch::config_t switch_config_a;
    switch_config_a.switch_cluster.current_position=0;
    switch_config_a.switch_cluster.number_of_positions=2;
    endpoint_t *switch_endpoint_a = generic_switch::create(node, &switch_config_a, ENDPOINT_FLAG_NONE, NULL);

    switch_endpoint_id_a = endpoint::get_id(switch_endpoint_a);     
  }
```

Any ideas? It feels like I'm missing some obvious step.

jakubdybczak, 2022-11-25:
Hey! Thank you for feedback!
Unfortunately I don't have Apple's hub to be able to test HomeKit, however I'm supposed to get update for my Google Nest soon so hopefully I would be able to reproduce these issues. I have tested connecting my ESP with CHIPTool only and I haven't noticed slow / unreliable connection, but I will keep an eye on this issue.

Regarding switch I would begin with testing it using CHIPTool -- official Matter tool for testing commisioning and controlling Matter devices - if it won't work there then most probably something is wrong with config.
Also I looked at `esp-matter` examples and could not find usage of `generic_switch`. The only example that I found is usage of `on_off_switch` there: https://github.com/espressif/esp-matter/blob/main/examples/light_switch/main/app_main.cpp. Maybe you can compare your config with this one? Also, recently I came across some graphics showing available device types in Apple's Home app - I don't know how reliable this source is but anyway will post it there.`OnOffSwitch` should work:
[image]

In next few days I will try to provide some more examples, including switch, tested with CHIPTool and (hopefully, if my Nest finally gets preview update) Google ecosystem.

scottgarner, 2022-11-25:
Thanks for the quick response! 

The generic switch type is confusing. I had seen some threads on other matter repos about on/off being used explicitly for lighting and generic being more general purpose, but I really can't find much more info about it. Given that the above chart doesn't even show the generic type, I'll go back to testing with on/off.

I wasn't having much luck with CHIPTool during initial testing, but I'll give that another shot, too. Thanks!

oleksiikutuzov, 2022-11-26:
I have a Philips Hue bridge, which enables Matter in beta. The dimmer switch is exposed to HomeKit via Matter and is working fine. According to developers, it is a "generic switch" cluster. So it definitely should be possible

scottgarner, 2022-11-27:
I did a bit more testing with an on/off switch without much luck, so I went back to a separate install of `esp-idf` and `esp-matter` to flash the `light_switch` example and had the same problem.

I posted an issue at espressif/esp-matter#98.

jakubdybczak, 2022-12-07:
Today I was able to add `generic_switch` to Google Home but I could not control it via app - it was only showing current state of switch. I have also tried controlling it via Google Nest, but is an information that "Touch controls are limited right now, but you can still use voice to interact with this device".

Yacubane, 2023-07-24:
Closing, because it's a Matter controller related problem

closed 2023-07-24 by Yacubane