Yacubane/esp32-arduino-matter #6: Matter.h: No such file or directory

LukasAV, 2022-12-06:
Hi,
I am getting the following error when trying to build. I feel like I am missing something obvious, but unfortunately I could not figure it out yet.
[image]
This is my project structure, I copied the `Light.ino` in the main.cpp, the library got cloned from git correctly
[image]
`platformio.ini`:
```
[env:nodemcu-32s]
platform = espressif32
board = esp32dev
framework = espidf
build_unflags=-std=gnu++11
build_flags=-std=gnu++17
lib_deps = https://github.com/jakubdybczak/esp32-arduino-matter.git
board_build.partitions = partitions_singleapp_large.csv
```

oleksiikutuzov, 2022-12-06:
I guess `framework` should be `arduino`. Mine works fine with this setup: https://github.com/oleksiikutuzov/esp32-matter-switch/blob/main/platformio.ini

LukasAV, 2022-12-06:
Thank you so much, I knew it was something stupid. I confused the framework with the platform.
Now it built without issues.

closed 2022-12-06 by LukasAV