Yacubane/esp32-arduino-matter #8: Can I implement to my own iot system

gerdal91, 2023-01-03:
Hello , I build an iot system for my products . I have my own application that uses Bluetooth for setup Wi-Fi credentials on esp32 and after that all the communication are over my server .

I want to implement matter to my product . But also I want to keep using my mobile application. 

I want my users to setup their Wi-Fi credentials over my mobile application.

I just want to open matter function over my mobile applications . Can I make something like that ? How can I ?

jakubdybczak, 2023-01-03:
Maybe you can just add additional section in your app with QR code / numeric code with which user can optionally setup your device as Matter device? But I'm not sure what you are trying to achieve. Do you want to have working both Matter and your custom protocol at once? I would also look at https://www.youtube.com/watch?v=ZSqhzTReAI8 that can enable control of Matter devices from your app but I guess that's not what you are looking for.

gkhnbks, 2023-01-03:
Yes, I want to have both working at same time.

for example, there is gonna be a button in my application . like add to matter. when user press this button my application will print the barcode for the device. 

so matter protocol can use the already saved wifi credentials by my mobile application to the device.

gerdal91, 2023-01-03:
Yes, I want to have both working at same time.

for example, there is gonna be a button in my application . like add to matter. when user press this button my application will print the barcode for the device.

so matter protocol can use the already saved wifi credentials by my mobile application to the device.

jakubdybczak, 2023-01-03:
>so matter protocol can use the already saved wifi credentials by my mobile application to the device

I'm not a developer of Matter so I can be wrong here but as far as a know there is not a way to do this. All popular Matter solutions works with use of hub, so you would need to pass additional information about hub in order to commision the device. Also commisioning process is not an easy task and it would be hard to implement it inside your application or directly on esp32. It's also platform depended. Have you seen how this process work on Google or Apple devices? During device commision your phone connects with Google services to add device to Google Fabric and this data is saved on cloud. I guess that you would have to analyze this code: https://github.com/project-chip/connectedhomeip/blob/66c0270ba88e026075c91988361cec0a70f7c7c8/src/platform/ESP32/NetworkCommissioningDriver.cpp#L152 (this snipped is for setupping wifi during commisioning). Maybe it will be easier to allow Matter commisioning process to change your default wifi credentials that user provided within app? I know that's not ideal from UX perspective but with good information it might work. During Matter commisioning process WiFi credentials are automatically taken from phone - currently connected WiFi credentials are passed to Matter device.

jakubdybczak, 2023-01-12:
I'm closing this issue as it's more related to https://github.com/project-chip/connectedhomeip. I do allow such discussions on this repository but I feel like this one covers very specific use case and is not very active. Feel free to write here if you find any information about this problem ;)

closed 2023-01-12 by jakubdybczak

HamzaHajeir, 2023-02-15:
Hi there, 
I'm interested also to run Matter with my under-developing system, which has a WiFi provisioning method over AP, custom communication security, webserver, and Mqtt communication management, and soon the OTA.

While I'm reading now, it's important to ensure that a some folks would be really having this kind of use-case, which including an investment of a developed system/framework at various shapes and at different stages.

My main concern is: 
- Does Matter SDK API take over the system completely? or there's a room for a collaborative/hybrid software?

Which is detailed to:
- Is there a potential of a WiFi system management by users (WiFi provisioning and manual mode) (Maybe a bridge a user could build).
- For Matter provisioning, is there a room to pass additional information as configuration (Some should be confidential)?
- Could someone develop customized OTA?
- Is the matter webserver collaborative to what someone could rely upon? (in terms of ports and Lwip/TCP usage).

Currently, I could envision a potential of a hybrid software, but with no crisp ideas.

Is there a simple, and yet informative description to Matter and Matter SDK you could recommend?

What do you think?

HamzaHajeir, 2023-02-21:
I think this should answer a lot of concerns:
https://github.com/espressif/esp-matter/issues/5