Yacubane/esp32-arduino-matter #12: Decommissioning

ruddycanuck, 2023-01-12:
I've been digging through this project, and the ESP/Connectedhomeip libraries, but I can't for the life of me figure out how to decommission my ESP32-C3. Well I can, I just have to upload the firmware again and that's not a great solution.
Is there a function somewhere that I'm missing that forces the ESP32 to either decommission or recommission?
Example: I have my prototype connected into CHIP, but I want to test it in Smartthings - I'd like to implement something like holding down a button while applying power to clear out the current pairing.

Or another use case is if Commissioning only works half way, the ESP32 thinks it is happy but the App cannot see it.

jakubdybczak, 2023-01-12:
Hey, I looked inside Matter repository and found:
`DeviceLayer::ConfigurationMgr().InitiateFactoryReset();`
It does compile but I haven't tested it on real device. Can you try it?

ruddycanuck, 2023-01-12:
YES! This works perfectly. I ended up using a second button instead of the 'holding down the button while applying power scheme' for testing.

closed 2023-01-12 by ruddycanuck