Yacubane/esp32-arduino-matter #14: Provisioning Time Out?

ruddycanuck, 2023-01-24:
ESP32-C3
I'm trying to figure out how to stop the ESP32-C3 from trying to provision after a set time. 

While provisioning the ESP32 is drawing around 90mA which heats up my vreg quite quickly. I could switch to a switching regulator but I'd also like to know how to stop the attempt.

Example - a freshly set up ESP32-C3 starts, tries to provision for 10 minutes, then gives up. The ESP32-C3 continues to operate with the io devices I have connected to it. I cycle power (or press a button or whatever) and it starts over trying to provision.

Is this implemented in your project or do I need to go to the more granular level / connectedhomeip ?

ruddycanuck, 2023-01-24:
I'm looking at simply stopping ble with ble_gap_adv_stop() but I woudl expect there to be a more graceful way to stop?

jakubdybczak, 2023-01-25:
Yes, you should look at connectedhomeip. This library is just a compiled wrapper of it. Unfortunately I don't know how to stop provisioning.

closed 2023-02-12 by jakubdybczak