Yacubane/esp32-arduino-matter #15: Connection with SmartThings Hub V3

Inscribe, 2023-02-03:
Thanks for this great work! I have been successful in commissioning the ESP32 switch with SmartThings, using an iPhone. It wasn't smooth and took many attempts, each time covering a few more steps before I got it connected. One major holdup was when it tried to write to iCloud and failed. I attributed it to my iCloud being full. But, then it went through. Stayed connected for two days.
Now the sad part, it has been offline since last 10 hours and I have tried power cycling the device. Not sure where to start investigating.
When it completed commissioning, Alexa (connected to SmartThings) picked it up immediately and Google Home, a few minutes later. Could control the switch using voice and app. Couldn't share with HA Matter server...couldn't figure out how.

Inscribe, 2023-02-03:
Good news....power cycled the hub and the device came back!

jakubdybczak, 2023-02-07:
>how to OTA update the ESP32
I would try https://lastminuteengineers.com/esp32-ota-updates-arduino-ide/
I haven't tested it with Matter, there is possibility that some commisioning data might be lost (I'm not sure where it's being stored, if in NVS, then it will work).

Regarding device being offline I have saw similar behavior with Google Hub, sometimes it just does not connect and I need to power cycle the ESP32 or hub. To be honest I don't know where the problem is :/

Inscribe, 2023-02-07:
OTA - Will wait for your feedback.
Offline - Resetting esp doesn’t seem to work. Resetting Hub always seems to work. Happens at night, always.