Yacubane/esp32-arduino-matter #17: Cannot locate esp_netif_types.h

daniel-frenkel, 2023-02-16:
Hello, I'm trying to compile the Light example and it fails immediately due to the following error

```
C:\Users\<USER>\Documents\Arduino\libraries\esp32-arduino-matter\src/platform/ESP32/CHIPDevicePlatformEvent.h:31:29: fatal error: esp_netif_types.h: No such file or directory
```

From my Google searches, this appears to be an [ESP-IDF file](https://github.com/espressif/esp-idf/blob/master/components/esp_netif/include/esp_netif_types.h) 

Is anyone else having this issue?

jakubdybczak, 2023-02-17:
Pls post your:

* ESP32 model
* Arduino version and version of board

daniel-frenkel, 2023-02-17:
I'm using the ESP32 Dev Module, however, it won't compile regardless of which ESP32 board that I select in Arduino so I don't think the board is the problem.

I am using Arduino 1.8.19 and the latest version of ESP32-arduino-matter

daniel-frenkel, 2023-02-17:
I think my board version is old. In the board manager it says I have 1.0.6 which I thought was the latest version, but looks like I'm wrong. You clearly state it needs to be 2.0.6 so I will fix that.

daniel-frenkel, 2023-02-17:
Everything works great, I just upgraded to ESP32 board version 2.0.6 and followed the rest of your instructions more carefully. Thanks for this amazing library

closed 2023-02-17 by daniel-frenkel

jakubdybczak, 2023-02-18:
It's sad that this lib is literally compiled against concrete version of ESP32 SDK. In perfect world we should compile Matter SDK within Arduino IDE, but Matter compilation process is so complicated and customized that it's not easy to port it into Arduino. Another idea would be to re-implement Matter for ESP32, but this task is not easy as well because Matter standard is quite... complicated, comparing it to other open standards, like Mozilla WebThings.