Yacubane/esp32-arduino-matter #19: Generic Switch States

nixmeer, 2023-02-21:
Hello :) 
I would like to create my own first matter firmware for the the [Sonoff M5]() which has 1 to 3 push buttons. To my understanding, the generic switch should be the right device type. However, I can't figure out which button states are allowed. My old wall switches supported single click, double click and long click which was also supported by Apple HomeKit. 

Best 

Jan

AintLarry, 2023-02-22:
Hey,

What exactly do you mean with button states? 

Maybe this link can help. It’s the official matter repo, so you can also find examples here.
https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/switch-server/switch-server.cpp

Another option is to write a simple script as base for your project, and read out the return output of the change-events.

Yacubane, 2023-07-24:
Closing due to inactivity. Let me know if you have still this problem.

closed 2023-07-24 by Yacubane