Yacubane/esp32-arduino-matter #22: compilation error

phil31, 2023-03-19:
Dear all, i try to compile the "light" example .

plateformio.ini is : 
[env:adafruit_qtpy_esp32c3]
platform = espressif32@6.1.0
board = adafruit_qtpy_esp32c3
framework = arduino
build_unflags = -std=gnu++11
build_flags = -std=gnu++17
board_build.partitions = min_spiffs.csv
imported lib matter in the directory lib/matter
compilation is working but not linking :
...
c:/users/phil/.platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch5/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: D:\WIFIPOWER\Matter\Matter/src/main.cpp:93: undefined reference to esp_matter::attribute::get_val(unsigned int*, esp_matter_attr_val_t*)' c:/users/phil/.platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch5/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: .pio\build\adafruit_qtpy_esp32c3\src\main.cpp.o: in function set_onoff_attribute_value(esp_matter_attr_val_t*)':
D:\WIFIPOWER\Matter\Matter/src/main.cpp:99: undefined reference to `esp_matter::attribute::update(unsigned short, unsigned long, unsigned long, esp_matter_attr_val_t*)'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\adafruit_qtpy_esp32c3\firmware.elf] Error 1

any advice please ?
thanks

jakubdybczak, 2023-03-19:
Hi, I tested it today and can confirm same problem. I'm looking at this.

jakubdybczak, 2023-03-20:
Actually, I did not look at this error correctly and I had different but quite similar error. I fixed this in [1.0.0-beta.3-branch](https://github.com/jakubdybczak/esp32-arduino-matter/tree/1.0.0-beta.3-branch), could you confirm if that works?

phil31, 2023-03-20:
Not realy better : 

 

 

 

Regards

Phil

jakubdybczak, 2023-03-21:
I just tried and compiled successfully light example with this config:

```
[env:esp32dev]
platform = espressif32@6.1.0
board = adafruit_qtpy_esp32c3
board_build.mcu = esp32c3
framework = arduino
build_unflags = -std=gnu++11
build_flags = -std=gnu++17
board_build.partitions = min_spiffs.csv
lib_deps = https://github.com/jakubdybczak/esp32-arduino-matter.git#1.0.0-beta.3-branch
monitor_speed = 115200
```

Could you send your project so I can debug it?

phil31, 2023-03-21:
Not working on my side, the only difference is that i import the matter lib directly in lib directory because i don’t have GIT on my PC.

 

You can download the project here : https://transfert.free.fr/thDsAIY

 

Thanks, regards

jakubdybczak, 2023-03-24:
I will look at this on weekend

jakubdybczak, 2023-03-25:
The problem is with your project structure, you copied `src` folder from this library to `lib/matter` folder. Proper structure is 
[image].

Tip: you can go to `lib` folder, clear the stuff here and type `git clone --depth 1 --branch 1.0.0-beta.3-branch https://github.com/jakubdybczak/esp32-arduino-matter` and that will be enough ;)

phil31, 2023-03-25:
Thanks Jakub, regards

 

 <https://user-images.githubusercontent.com/37028087/227713272-a77c71f3-fa60-429b-9b99-94efa5833834.png> .

closed 2023-03-25 by jakubdybczak