Yacubane/esp32-arduino-matter #23: Current Build not working due to issue in ESP-Matter Lib

weidingerc, 2023-03-27:
Due to this issue https://github.com/espressif/esp-matter/issues/217 , which already has been fixed, the Matter device with an ESP32 cannot connect to an iPhone with Apple Home. I used the Demo Example Code named "Light", unfortunately with the above described bug.

Could you recompile this lib to fix this bug? 
Would be awesome :-)

jakubdybczak, 2023-03-28:
I'm working on it ;)

jakubdybczak, 2023-03-29:
If anyone needs it right now, I'm sending build via Google Drive: https://drive.google.com/drive/folders/1sKJHu-eRWWtuSjzGn-l8Wf4U2QkKAREE?usp=sharing

I need to rethink structure of this repository, as compiled files are too large. Most probably I will merge this and https://github.com/jakubdybczak/esp32-arduino-matter-builder repository, and repository will not have compiled files. Compiled files will be only as attached files in github packages. From the other hand this approach is not really good for platformio developers, where it's easier to type repository url and in this case git-lfs is also an option.

weidingerc, 2023-03-30:
Thanks for the quick answer! Unfortunately when I now try the light example, it still cannot connect and the Phone shows that it cannot add the device :-(

it seems it still cannot connect to my network (I was also making sure be be in the 2.4GHz WLAN before setup). I'll have to try with another router. But currently my home setup is then probably not supported then.

D (68352) chip[DMG]: Received command for Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0006
I (68358) chip[NP]: ESP NetworkCommissioningDelegate: SSID: WLAN-W Extended
I (68368) chip[DL]: WiFi station mode change: Enabled -> Disabled
I (68390) chip[DL]: Changing ESP WiFi mode: STA -> STA+AP
I (68395) chip[DL]: WiFi station mode change: Disabled -> Enabled
D (68395) chip[DMG]: Decreasing reference count for CommandHandler, remaining 1
I (68399) chip[DL]: Attempting to connect WiFi station interface
I (68405) chip[DL]: WiFi station state change: NotConnected -> Connecting
I (68409) chip[DL]: Done driving station state, nothing else to do...
I (68417) chip[DL]: Attempting to connect WiFi station interface
E (68421) wifi:sta is connecting, return error
E (68427) chip[DL]: esp_wifi_connect() failed: ESP_ERR_WIFI_CONN
I (68433) chip[DL]: Attempting to connect WiFi station interface
E (68437) wifi:sta is connecting, return error
E (68442) chip[DL]: esp_wifi_connect() failed: ESP_ERR_WIFI_CONN
I (68446) chip[DL]: WIFI_EVENT_AP_START
I (68450) chip[DL]: WiFi AP state change: NotActive -> Active
I (68456) chip[DL]: Changing ESP WiFi mode: STA+AP -> STA
I (68465) chip[DL]: WiFi AP state change: Active -> Deactivating
E (68469) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
I (68475) chip[DL]: WIFI_EVENT_AP_STOP
I (68479) chip[DL]: WiFi AP state change: Deactivating -> NotActive
I (70349) chip[DL]: WIFI_EVENT_STA_CONNECTED
I (70349) chip[DL]: WiFi station state change: Connecting -> Connecting_Succeeded
I (70351) chip[DL]: WiFi station state change: Connecting_Succeeded -> Connected
I (70357) chip[DL]: WiFi station interface connected

diysmartmatter, 2023-03-30:
> If anyone needs it right now, I'm sending build via Google Drive: https://drive.google.com/drive/folders/1sKJHu-eRWWtuSjzGn-l8Wf4U2QkKAREE?usp=sharing

Thank you very much! The new build on the Google Drive worked fine.

diysmartmatter, 2023-03-30:
> it still cannot connect and the Phone shows that it cannot add the device :-(

You may try to update Apple Home from "Home Settings" menu at the top right corner to activate Matter. It seems that the update is not automatic as it may cause incompatibility among shared users and computers.

weidingerc, 2023-03-30:
I already updated to the new “Home” architecture on iOS. So I guess it is another problem
Are you trying to add the esp32 matter device also with an iOS device?

diysmartmatter, 2023-03-30:
I am running the light example on my iPhone with iOS 16.4. I also have a HomePod mini (16.4) working as a home hub. 

https://user-images.githubusercontent.com/124488991/228818068-660993ec-cb3a-4779-883f-dec159e85192.mov

weidingerc, 2023-03-30:
Yay, I got it working by adding it my iPad. But I honestly have no idea why it only worked on this device. Now I can start coding :-)
Thanks for the help and compilation

closed 2023-03-30 by weidingerc