Yacubane/esp32-arduino-matter #24: WindowCovering Example

weidingerc, 2023-03-31:
Hi there, has anyone experience how to setup a window covering device.
I tried to modify the LightOnOff Example, but the device always shows as "Not responding".
I also could not find any example online.

daniel-frenkel, 2023-05-12:
Have you figured it out?

weidingerc, 2023-05-15:
Unfortunately not

Yacubane, 2023-05-15:
I may look at this. Which Matter Controller are you using?

weidingerc, 2023-05-15:
HomePod mini as Border Router
ESP32 as controller (https://www.amazon.de/dp/B071P98VTG?psc=1&ref=ppx_yo2ov_dt_b_product_details)

Sustainable-Me, 2023-11-02:
Our window shade controller just bit the big one and i was going to replace it with a matter/eps32 by modifying the LightOnOff Example (Matter project # 3 for me :)  ) through google home wifi.  Anybody have luck moving this forward?

weidingerc, 2023-11-26:
Any luck getting it to work?

Robothaler, 2023-12-01:
**@weidingerc** 
The hint worked with my code, the Matter device is not marked with "Not responding".
You will find this already incorporated in my code.
**https://github.com/espressif/esp-matter/issues/343#issuecomment-1531494497**

Unfortunately my programming skills are not good, so I have difficulties to finalize my project.

I want to control belt winders, but I don't know how to redirect the Matter functions to my code.
Maybe someone can give me a push in the right direction.

Here is my current code:
**https://github.com/Robothaler/BeltWinder-Matter/tree/dev**

weidingerc, 2023-12-03:
@Robothaler Thanks a lot, I got it working with the additional control clusters in your code. 
But what I still did not manage was to just have a "Default lifting" window covering that just can "open/close" and nothing more.

Robothaler, 2023-12-03:
**@weidingerc**I'm pleased that you're now one step further.
With my code, the Apple Home and EVE app only show me a roller shutter with up/down and stop function.
You may also be interested in how the WindowCovering type can be changed.
Take a look at the following entry:
**https://github.com/espressif/esp-matter/issues/280#issuecomment-1488907563**

The required line is the following:
`window_covering_device::config_t wcd_config(static_cast<uint8_t>(chip::app::Clusters::WindowCovering::EndProductType::kRollerShutter));`

How do you control your roller shutter? In other words, how do you transfer the position change to your code? Frankly, I have not yet understood the Matter functions.
In my code, I want to pass the new position from the Home app to my `newPosition` function to the `newPercentage` variable. Maybe you can help me at this point.
Also, I can't see any custom DEBUG output in the serial output.
I have already changed this to the following format:

`ESP_LOGD(TAG, "window_covering_device cluster created with cluster_id %d", cluster);`
I still can't see the output!

weidingerc, 2023-12-03:
@Robothaler Thanks, I‘ll try this different window covering config once I have time again ;-P

weidingerc, 2023-12-03:
> **@weidingerc**I'm pleased that you're now one step further. With my code, the Apple Home and EVE app only show me a roller shutter with up/down and stop function. You may also be interested in how the WindowCovering type can be changed. Take a look at the following entry: **[espressif/esp-matter#280 (comment)](https://github.com/espressif/esp-matter/issues/280#issuecomment-1488907563)**
> 
> The required line is the following: `window_covering_device::config_t wcd_config(static_cast<uint8_t>(chip::app::Clusters::WindowCovering::EndProductType::kRollerShutter));`
> 
> How do you control your roller shutter? In other words, how do you transfer the position change to your code? Frankly, I have not yet understood the Matter functions. In my code, I want to pass the new position from the Home app to my `newPosition` function to the `newPercentage` variable. Maybe you can help me at this point. Also, I can't see any custom DEBUG output in the serial output. I have already changed this to the following format:
> 
> `ESP_LOGD(TAG, "window_covering_device cluster created with cluster_id %d", cluster);` I still can't see the output!

This line you suggested only leads to "Device not responding" again. I guess I'll stay at the Glider shades from 0 to 100% to open and close my shades.

Robothaler, 2023-12-26:
> I have also had some luck with the matter window covering device -- A simple system controlling multiple shades, one at a time, using a single esp32, a timer for the intermediate shade positions, and limit switches for full open/close. I'm not a programmer, I only play one in retirement ;)
> 
> I haven't had a chance to work carefully through the recent comments, yet for me stumbling upon the "int val->val.i" in the static esp_err_t on_attribute_update routine was the key moving from the "bool val->val.b" in the two existing on/off device examples to a variable slider for the shade positions. val.i returns the % closed (which is 10000 - the % open value in the google home slider). Will try the "val->val.u16" as suggested as that seems more logical and should match the GH slide.
> 
> Neither the get- or set- attributes, nor the matter shade functions are working for me either - it would be great to learn more about the matter shade controller's inner workings...
> 
> However, The current problem I need to solve for intermediate shade positions before implementation is: when moving the slider slowly in the GH app, I am getting multiple values for the open % (val->val.i) along the way. Thus, moving from say 20% to 80% open is returning intermediate values such as 42%, 57%, and 67% before my finger stops moving the slider at 80%. It shouldn't be too hard to figure this out with a simple timer.
> 
> Having said that, I wonder if there is a matter function out there that takes care of this automatically?? For full open/close of the shades, the limit switches work.
> 
> I thought it might be useful to attach my minimalist Arduino ide code skeleton which commissions a two shade slider group in GH. Perhaps the experts can chime in what matter functions are missing and how to use them... [Minimal_window_covering.zip](https://github.com/Yacubane/esp32-arduino-matter/files/13539138/Minimal_window_covering.zip)

**@Sustainable-Me** I have solved the problem of intermediate values when dragging the slider with a new function. The incoming values are buffered there and forwarded to the function for determining the position after a filter time. This works quite well for my case. Here is my current code, of course you still have to create the appropriate global variables.

` // buffering new target values
    void bufferNewPosition() {
    unsigned long currentTime = millis();

    // The very first new target value received
    if (newPercentageReceived && !bufferedValueSaved) {
        bufferedPercentage = newPercentage;
        DEBUGPRINTLN2("[BELTWINDER] bufferNewPosition: New percentage received - " + String(newPercentage) + "%");
        DEBUGPRINTLN2("[BELTWINDER] bufferNewPosition: First value buffered - " + String(bufferedPercentage) + "%");

        bufferedValueSaved = true;
        
        lastUpdateTime = currentTime;
        DEBUGPRINTLN2("[BELTWINDER] lastUpdateTime was 0 will be changed to: New timestamp - " + String(lastUpdateTime) + "ms");
        return;
        }
        
    // A new target value received within filter duration
    if (newPercentageReceived && bufferedValueSaved && currentTime - lastUpdateTime <= filterDuration) {

      if (newPercentage == bufferedPercentage) {
        DEBUGPRINTLN2("[BELTWINDER] bufferNewPosition: New percentage is equal to buffered Percentage - " + String(newPercentage) + "%");
      } else {
        bufferedPercentage = newPercentage;
        bufferedValueSaved = true;
        return; 
        }
    } else if (newPercentageReceived && bufferedValueSaved && currentTime - lastUpdateTime > filterDuration){
      // Time has expired, use the last buffered value
        DEBUGPRINTLN2("[BELTWINDER] bufferNewPosition: Time limit for buffering exceeded. Use last buffered value for new position.");
        filteredPercentage = bufferedPercentage;
        
        DEBUGPRINTLN2("[BELTWINDER] bufferNewPosition: Filtered percentage - " + String(filteredPercentage));

        lastUpdateTime = 0; // Set lastUpdateTime to 0 for the next update
        newPercentageReceived = false;
        bufferedValueSaved = false;
        newPosition(filteredPercentage);
    }
}`

palsbo, 2024-02-17:
I made a Matter device with 3 windowCovering endpoints. It controls 3 Somfy blinds and I can open, close at set to x% opening from Google Home. It is build using Arduino IDE with esp32_arduino_matter library. 

It works fine, but II do have a few problems. The conversion from the matter device to Google Home (and other apps) "Up" to be position 100% and down  to be 0%.  So when I want to position to 20% open, I get 80% open.
I tried to do a manual convertion by calculating the opening percentage top 100-value. This works, but on power up, all blinds go down to fully extended position. not good en a power failure situation when the storm is on!.

Either I need to invert the action between the matter device and google home, or I need to be able to set start position to fully extended and then calculate 100-value.

Any idea is apreciated.
Regards
erik

Robothaler, 2024-02-18:
@palsbo 
Hi,
you can find the solution in my repository:
[BeltWinder-Matter)](https://github.com/Robothaler/BeltWinder-Matter/tree/main/src)

You have to use "Mode" and change the motordirection:

```
// Attribute: Id 23 Mode
  set_window_covering_mode(window_covering_endpoint_id, Mode::kMotorDirectionReversed);
```

To submit the new mode I've created this function:

```
void set_window_covering_mode(chip::EndpointId endpoint, chip::app::Clusters::WindowCovering::Mode newMode)
{
    esp_matter_attr_val_t modeValue;
    modeValue = esp_matter_int8(static_cast<int8_t>(newMode));

    esp_err_t result = attribute::update(window_covering_endpoint_id, WINDOW_COVERING_CLUSTER_ID, WINDOW_COVERING_ATTRIBUTE_ID_MODE, &modeValue);

    if (result == ESP_OK)
    {
        DEBUGPRINTLN2("[BELTWINDER] Window covering Mode updated successfully");
        chip::DeviceLayer::PlatformMgr().LockChipStack();
        DEBUGPRINTLN2("[BELTWINDER] Current Mode: "); ModePrint(ModeGet(window_covering_endpoint_id));
        chip::DeviceLayer::PlatformMgr().UnlockChipStack();
    }
    else
    {
        DEBUGPRINTLN2("[BELTWINDER] Failed to update window covering Mode");
        // Handle error if needed
    }
}
```

I hope this will help you.

palsbo, 2024-02-18:
Thank your for quick response. I will try  it.
Best regards
Erik