Yacubane/esp32-arduino-matter #25: How to obtain pin and discriminator / full device boot log?

malyjak, 2023-04-12:
Sorry for having such a trivial question but I'm unable to obtain these values. When I used Espressif sdk in the past, I could see all device information at boot, including the pin and discriminator. How do I enable this in the code, so I can see the same log once the platformio flashes the buit firmware and esp boots up?

malyjak, 2023-04-13:
Update to more precisely formulate the question: Having 'ESP_LOG_DEBUG' set, one can see the discriminator, qr code url, and manual pairing code. However, there is no PIN info. Using the default `20202021` works well to pair through the chip-tool. So this question should be more focused on why the log differs from the default Espressif sdk.

The default sdk log:
```
I: 490 [DL]  Setup Pin Code (0 for UNKNOWN/ERROR): 20202021
I: 495 [DL]  Setup Discriminator (0xFFFF for UNKNOWN/ERROR): 3840 (0xF00)
```

Light demo code log:
```
I (1777) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1
```

Since the PIN can't be changed as noted in readme, this is not an important issue, as long as the default value works. However, it would be nice to see the default value in logs.

Yacubane, 2023-05-14:
In newest version of this lib (beta 5) there should be a possibility to change this PIN. Please look here:
https://github.com/espressif/esp-matter/blob/4707b88ceb5e6328f7b571b652487507429c1891/docs/en/developing.rst (`2.5 Factory Data Providers` section)

I have enabled custom providers for every type of provider in that build.

I know that this does not really answer the question, but I still haven't yet discovered a way to print actual PIN code ;)