Yacubane/esp32-arduino-matter #27: [Error] Matter and WiFi connectivity

MiguelGomesEstevao, 2023-04-21:
I'm currently trying to send data from my ESP32 to Google Home using the Matter Protocol.
I managed to get it on my Google Home app but it appears as offline, so I tried to connect the device using the WiFi library so it can send data do the endpoint (Don't know if that helps, I was hoping to get some insight about that too).

However, this error message appears on my serial monitor log when I run the code.
I've been trying to fix this for days. Is the WiFi loop interfering with the creation of the endpoint somehow?
I'm sending my code below. I would be glad if someone could help me. Thank you in advance.

[Sensor.zip](https://github.com/Yacubane/esp32-arduino-matter/files/11298331/Sensor.zip)

**ERROR MESSAGE:**

```Connecting to WiFi...
WiFi connected
E (3140) esp_matter_core: Error create default event loop

abort() was called at PC 0x4015674d on core 1

Backtrace: 0x40083769:0x3ffd7320 0x40093e19:0x3ffd7340 0x40099919:0x3ffd7360 0x4015674d:0x3ffd73e0 0x40156246:0x3ffd7400 0x401565f9:0x3ffd7430 0x400d255c:0x3ffd7460 0x400d25eb:0x3ffd74e0 0x400d546e:0x3ffd7510

ELF file SHA256: c5681ca225aa6b52

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0```

Yacubane, 2023-04-25:
You should not connect Matter devices with WiFi library - Matter library should take care of it and reconnect automatically. I think that you should debug logs from Matter.

MiguelGomesEstevao, 2023-04-25:
I got to define the ssid and password somewhere in my code right? 
I've been searching the source code and the examples but I can't find any example of that.
Im able to detect my matter device and add it to my Google Home App but it says my device is offline.

Yacubane, 2023-04-25:
During pairing, credentials of WiFi where you are currently connected are send from your phone to Matter device. Matter then uses it to join network. Maybe your network is causing problems? Try changing SSID or network to 2.4GHz. I would also look through logs carefully - there should be info about trying to connect to WiFi during pair process

MiguelGomesEstevao, 2023-04-26:
You were right, it is able to pair the WiFi. 

The problem now is that the attribute::update is not being able to update the endpoint.
Do you know what might be causing this?
Sorry for making lots of questions, I really need to implement this for a work.
I think I configured everything well looking at the examples that are available.

I'll send my code below in case of need. Thank you in advance.

[MatterProtocol.zip](https://github.com/Yacubane/esp32-arduino-matter/files/11335239/MatterProtocol.zip)

`
E (161758) esp_matter_attribute: esp_matter_attr_val_type_t not handled: 0
E (161758) esp_matter_attribute: Could not allocate value buffer`

meta-space, 2023-05-24:
> During pairing, credentials of WiFi where you are currently connected are send from your phone to Matter device. Matter then uses it to join network. [...]

Thank you for that insight. Does this mean that ESP-Matter stack cannot share connections with other services? I'm currently building a commercial product which supports WiFi and Ethernet connections. On the application level we're supporting multiple automation protocols.  We were planning on configuring the network ourselves. But now I'm thinking that might not be possible with Matter...