Yacubane/esp32-arduino-matter #28: Example usage

ba1yuu, 2023-04-23:
Hi, i am a student who wants to try to connect matter with esp32 to control leds, but I saw that your sample program cannot be connected. I would like to ask if there is a backup.

Nickwren, 2023-08-30:
Im also finding the example no longer works. I have tried enabling the device on google and alexa. Every time it seems to work, accepting the qr code then asking for confirmation that your adding a uncertified device and then it fails.

ba1yuu, 2023-09-05:
You can try testing on Apple devices. It seems that Google and Alexa require product certification before use, otherwise, you may encounter authentication issues. Currently, I am switching to a Linux environment for testing.