Yacubane/esp32-arduino-matter #30: Support OTA by default / add switch

malyjak, 2023-04-24:
As noted [here](https://docs.espressif.com/projects/esp-matter/en/main/esp32/production.html#over-the-air-ota-updates) the Espressif's Matter SDK supports ota requestor out of the box. However, after announcing the provider via chip tool to the ESP32 running the light app code based on this library, an unsupported cluster error emerges:

```
>>> otasoftwareupdaterequestor announce-ota-provider 12345 0 0 0 0x7283 0
[1682327383.313946][3064:3069] CHIP:TOO: Sending command to node 0x7283
[1682327383.314487][3064:3069] CHIP:CSM: FindOrEstablishSession: PeerId = [1:0000000000007283]
[1682327383.314543][3064:3069] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1682327383.314592][3064:3069] CHIP:CTL: Found an existing secure session to [1:0000000000007283]!
[1682327383.314640][3064:3069] CHIP:CTL: OperationalSessionSetup[1:0000000000007283]: State change 1 --> 5
[1682327383.314688][3064:3069] CHIP:TOO: Sending cluster (0x0000002A) command (0x00000000) on endpoint 0
[1682327383.314748][3064:3069] CHIP:DMG: ICR moving to [AddingComm]
[1682327383.314797][3064:3069] CHIP:DMG: ICR moving to [AddedComma]
[1682327383.314958][3064:3069] CHIP:EM: <<< [E:53616i M:146168965] (S) Msg TX to 1:0000000000007283 [015A] --- Type 0001:08 (IM:InvokeCommandRequest)
[1682327383.315024][3064:3069] CHIP:IN: (S) Sending msg 146168965 on secure session with LSID: 40899
[1682327383.315324][3064:3069] CHIP:DMG: ICR moving to [CommandSen]
[1682327383.507389][3064:3069] CHIP:EM: >>> [E:53616i M:122838963 (Ack:146168965)] (S) Msg RX from 1:0000000000007283 [015A] --- Type 0001:09 (IM:InvokeCommandResponse)
[1682327383.507464][3064:3069] CHIP:EM: Found matching exchange: 53616i, Delegate: 0x7fa0031d18
[1682327383.507516][3064:3069] CHIP:EM: Rxd Ack; Removing MessageCounter:146168965 from Retrans Table on exchange 53616i
[1682327383.507568][3064:3069] CHIP:DMG: ICR moving to [ResponseRe]
[1682327383.507631][3064:3069] CHIP:DMG: InvokeResponseMessage =
[1682327383.507676][3064:3069] CHIP:DMG: {
[1682327383.507721][3064:3069] CHIP:DMG:        suppressResponse = false,
[1682327383.507774][3064:3069] CHIP:DMG:        InvokeResponseIBs =
[1682327383.507830][3064:3069] CHIP:DMG:        [
[1682327383.507881][3064:3069] CHIP:DMG:                InvokeResponseIB =
[1682327383.507946][3064:3069] CHIP:DMG:                {
[1682327383.508007][3064:3069] CHIP:DMG:                        CommandStatusIB =
[1682327383.508077][3064:3069] CHIP:DMG:                        {
[1682327383.508146][3064:3069] CHIP:DMG:                                CommandPathIB =
[1682327383.508224][3064:3069] CHIP:DMG:                                {
[1682327383.508304][3064:3069] CHIP:DMG:                                        EndpointId = 0x0,
[1682327383.508392][3064:3069] CHIP:DMG:                                        ClusterId = 0x2a,
[1682327383.508480][3064:3069] CHIP:DMG:                                        CommandId = 0x0,
[1682327383.508566][3064:3069] CHIP:DMG:                                },
[1682327383.508647][3064:3069] CHIP:DMG:
[1682327383.508724][3064:3069] CHIP:DMG:                                StatusIB =
[1682327383.508805][3064:3069] CHIP:DMG:                                {
[1682327383.508886][3064:3069] CHIP:DMG:                                        status = 0xc3 (UNSUPPORTED_CLUSTER),
[1682327383.508972][3064:3069] CHIP:DMG:                                },
[1682327383.509051][3064:3069] CHIP:DMG:
[1682327383.509187][3064:3069] CHIP:DMG:                        },
[1682327383.509262][3064:3069] CHIP:DMG:
[1682327383.509330][3064:3069] CHIP:DMG:                },
[1682327383.509392][3064:3069] CHIP:DMG:
[1682327383.509451][3064:3069] CHIP:DMG:        ],
[1682327383.509508][3064:3069] CHIP:DMG:
[1682327383.509560][3064:3069] CHIP:DMG:        InteractionModelRevision = 1
[1682327383.509611][3064:3069] CHIP:DMG: },
[1682327383.509674][3064:3069] CHIP:DMG: Received Command Response Status for Endpoint=0 Cluster=0x0000_002A Command=0x0000_0000 Status=0xc3
[1682327383.509731][3064:3069] CHIP:TOO: Error: IM Error 0x000005C3: General error: 0xc3 (UNSUPPORTED_CLUSTER)
[1682327383.509784][3064:3069] CHIP:DMG: ICR moving to [AwaitingDe]
[1682327383.509898][3064:3069] CHIP:EM: <<< [E:53616i M:146168966 (Ack:122838963)] (S) Msg TX to 1:0000000000007283 [015A] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1682327383.509973][3064:3069] CHIP:IN: (S) Sending msg 146168966 on secure session with LSID: 40899
[1682327383.510054][3064:3064] CHIP:TOO: Run command failure: IM Error 0x000005C3: General error: 0xc3 (UNSUPPORTED_C
```

My partitions.csv:
```
# Name,   Type, SubType, Offset,   Size
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,      data, nvs,     ,         20K
otadata,  data, ota,     ,         8K
ota_0,    0,    ota_0,   ,         1792K
ota_1,    0,    ota_1,   ,         1792K
spiffs,   data, spiffs,  ,         384K
coredump, data, coredump,,         64K
```

Commands prior to the announcement:
```
chip-tool interactive start
pairing ble-wifi 0x7283 WIFI_SSID WIFI_PSK 20202021 3840
onoff toggle 0x7283 0x1

--- In a new ssh terminal at the same device: chip-ota-provider-app -f update.bin

pairing onnetwork 12345 20202021
accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": null, "targets": [{"cluster": 41, "endpoint": null, "deviceType": null}]}]' 12345 0
otasoftwareupdaterequestor announce-ota-provider 12345 0 0 0 0x7283 0
```

Yacubane, 2023-04-25:
I think that I missed adding `CHIP_DEVICE_CONFIG_ENABLE_OTA_REQUESTOR` define during compilation, I will look into that

Yacubane, 2023-05-13:
Actually, it should work, because in every default esp-matter it's turned on. I started working on migrating new esp-matter version and will look at this once again.

Yacubane, 2023-05-14:
Okay, so it turned out that not every `esp-matter` example ships with OTA. I tried turning it ON it newest version of this lib (`beta5`). However, I haven't tested it.

malyjak, 2023-05-17:
Short story: It WORKS!

Long story:

ESP's requestor seemed to be working:
```
I (517134) chip[SWU]: OTA Requestor received AnnounceOTAProvider
D (517139) chip[SWU]:   FabricIndex: 1
D (517143) chip[SWU]:   ProviderNodeID: 0x0000000000003039
D (517148) chip[SWU]:   VendorID: 0x0
D (517151) chip[SWU]:   AnnouncementReason: 0
D (517155) chip[SWU]:   Endpoint: 0
```

However, linux provider failed at this point:
```
[1684331556.146219][4117:4117] CHIP:DMG: InvokeRequestMessage =
[1684331556.146267][4117:4117] CHIP:DMG: {
[1684331556.146311][4117:4117] CHIP:DMG:        suppressResponse = false,
[1684331556.146365][4117:4117] CHIP:DMG:        timedRequest = false,
[1684331556.146416][4117:4117] CHIP:DMG:        InvokeRequests =
[1684331556.146474][4117:4117] CHIP:DMG:        [
[1684331556.146524][4117:4117] CHIP:DMG:                CommandDataIB =
[1684331556.146578][4117:4117] CHIP:DMG:                {
[1684331556.146632][4117:4117] CHIP:DMG:                        CommandPathIB =
[1684331556.146694][4117:4117] CHIP:DMG:                        {
[1684331556.146758][4117:4117] CHIP:DMG:                                EndpointId = 0x0,
[1684331556.146826][4117:4117] CHIP:DMG:                                ClusterId = 0x29,
[1684331556.146892][4117:4117] CHIP:DMG:                                CommandId = 0x0,
[1684331556.146956][4117:4117] CHIP:DMG:                        },
[1684331556.147018][4117:4117] CHIP:DMG:
[1684331556.147074][4117:4117] CHIP:DMG:                        CommandFields =
[1684331556.147135][4117:4117] CHIP:DMG:                        {
[1684331556.147200][4117:4117] CHIP:DMG:                                0x0 = 65521,
[1684331556.147268][4117:4117] CHIP:DMG:                                0x1 = 32768,
[1684331556.147335][4117:4117] CHIP:DMG:                                0x2 = 1,
[1684331556.147402][4117:4117] CHIP:DMG:                                0x3 = [
[1684331556.147470][4117:4117] CHIP:DMG:                                                0,
[1684331556.147544][4117:4117] CHIP:DMG:                                ],
[1684331556.147612][4117:4117] CHIP:DMG:                                0x4 = 0,
[1684331556.147682][4117:4117] CHIP:DMG:                                0x5 = "XX" (2 chars),
[1684331556.147751][4117:4117] CHIP:DMG:                                0x6 = false,
[1684331556.147816][4117:4117] CHIP:DMG:                        },
[1684331556.147876][4117:4117] CHIP:DMG:                },
[1684331556.147935][4117:4117] CHIP:DMG:
[1684331556.147984][4117:4117] CHIP:DMG:        ],
[1684331556.148041][4117:4117] CHIP:DMG:
[1684331556.148088][4117:4117] CHIP:DMG:        InteractionModelRevision = 1
[1684331556.148138][4117:4117] CHIP:DMG: },
[1684331556.148210][4117:4117] CHIP:DMG: AccessControl: checking f=1 a=c s=0x0000000000007283 t= c=0x0000_0029 e=0 p=o
[1684331556.148270][4117:4117] CHIP:DMG: AccessControl: allowed
[1684331556.148316][4117:4117] CHIP:DMG: Received command for Endpoint=0 Cluster=0x0000_0029 Command=0x0000_0000
[1684331556.148377][4117:4117] CHIP:ZCL: OTA Provider received QueryImage
[1684331556.148419][4117:4117] CHIP:ZCL:   VendorID: 0xfff1
[1684331556.148458][4117:4117] CHIP:ZCL:   ProductID: 32768
[1684331556.148498][4117:4117] CHIP:ZCL:   SoftwareVersion: 1
[1684331556.148538][4117:4117] CHIP:ZCL:   ProtocolsSupported: [
[1684331556.148629][4117:4117] CHIP:ZCL:     0
[1684331556.148673][4117:4117] CHIP:ZCL:   ]
[1684331556.148712][4117:4117] CHIP:ZCL:   HardwareVersion: 0
[1684331556.148752][4117:4117] CHIP:ZCL:   Location: XX
[1684331556.148792][4117:4117] CHIP:ZCL:   RequestorCanConsent: 0
[1684331556.152805][4117:4117] CHIP:SWU: Error parsing OTA image header: ../../third_party/connectedhomeip/src/lib/core/OTAImageHeader.cpp:112: CHIP Error 0x000000DA: The file identifier, encoded in the first few bytes of a processed file, has unexpected value
[1684331556.152908][4117:4117] CHIP:SPT: VerifyOrDie failure at ../../third_party/connectedhomeip/examples/ota-provider-app/ota-provider-common/OTAProviderExample.cpp:359: ParseOTAHeader(parser, mOTAFilePath, header) == true
```

This error was provided while using the `firmware.bin` from `.pio\build\esp32dev` as the new firmware directly.
After running it through the `ota_image_tool.py` as described [HERE](https://developers.home.google.com/matter/ota/configuration) and using it, the OTA udate was successful!

Question/request:
Do you think it'd be possible to generate one more binary, called for example `firmware_ota.bin` in the post-compilation step via the `ota_image_tool.py` automatically?
PS: It would be great to modify arguments like `versionNumber` and `versionString` for this script through `platfromio.ini` because if it stays the same, matter requestor will ignore the update:
```
D (297196) chip[SWU]: Available update version 1 is <= current version 1, update ignored
```

Proposal:
I can summarize my findings and help create a new doc for this project for using OTA if you want ;)

Sustainable-Me, 2023-11-02:
Any luck/progress on the OTA documentation?   I would be more than happy to be a tester, proof reader, guinea pig, newbie sounding board, etc.  if that will help.

malyjak, 2023-11-08:
Hi. I stopped working with matter and completely forgot about my promise here. Basically you need chip-tool running on the same network. I highly suggest to use Linux as standalone PC or virtual machine. Then you need to follow several steps that enclosed in the bottom of the first post.

To summarize:
1) Do wifi-ble commisioning of your ESP device though the chip-tool
2) Try that LED on your device can be toggle via matter network (also through the chip-tool)
3) Start chip-ota-provider-app in a new terminal / as a background process
4) Set correct permissions for OTA provider
5) Observe your ESP device being updated

Note:
You need to use ota_image_tool.py provided by matter to properly pack your .bin files for OTA update via matter network