Yacubane/esp32-arduino-matter #31: Explain how to configure PlatformIO in more detail

richievos, 2023-04-25:
Supporting PlatformIO is great, but the current instructions aren't super clear on what you're supposed to do. In general a full example project would be super helpful, and/or a platformio.ini example.

Some specifics:

1. Use espressif32 platform at version v6.1.0 
    * would be great to show how to pin that
2. Turn on C++17 support, by setting build_unflags=-std=gnu++11 and build_flags=-std=gnu++17
    * relatively straightforward
4. [Download](https://github.com/jakubdybczak/esp32-arduino-matter/releases) and put library into lib folder of project (⚠️ you cannot use lib_deps in platformio.ini, because this repository does not contain binaries due to too big size)
    *  it's not clear what you're supposed to put in the lib folder. Eg do you put the entire contents of the folder, or just the src folder, or ...
    * I'm 90% sure it's the entire folder
5. Choose larger partition scheme, for example board_build.partitions=min_spiffs.csv
    * would be helpful to know what that config does, and where someone finds various csv options, and if that file is actually needed to exist on the filesystem or if it's a magic string
6. Run example sketch
    * straightforward

I'll volunteer to document that once I figure it out, but I'm currently not super clear what the steps are supposed to be, and I'm not clear if my setup is causing #29 or if I'm setting it up properly.

richievos, 2023-04-26:
For the next person, in the meantime:

```
[env:esp32devarduino]
platform = espressif32@6.1.0
board = esp32dev
framework = arduino

board_build.partitions=min_spiffs.csv

build_unflags =
    '-std=gnu++11'

build_flags =
    '-std=gnu++17'
```

Yacubane, 2023-05-13:
I added more info about platformio setup in readme.md

closed 2023-05-13 by Yacubane