Yacubane/esp32-arduino-matter #33: Light.ino example is tripping watchdog timeout in ESP32

meta-space, 2023-05-25:
I'm trying to get the library to run on my devices. But I get watchdog errors.
For instance running the Light.ino on my M5Stack Core (ESP32) I get:

```
I (100) wifi:wifi driver task: 3ffdc500, prio:23, stack:6656, core=0
I (993) system_api: Base MAC address is not set
I (998) system_api: read default base MAC address from EFUSE
I (1023) wifi:wifi firmware version: 6567a16
I (1023) wifi:wifi certification version: v7.0
I (1024) wifi:config NVS flash: enabled
I (1024) wifi:config nano formating: disabled
I (1028) wifi:Init data frame dynamic rx buffer num: 32
I (1033) wifi:Init management frame dynamic rx buffer num: 32
I (1038) wifi:Init management short buffer num: 32
I (1043) wifi:Init static tx buffer num: 8
I (1047) wifi:Init static rx buffer size: 2212
I (1051) wifi:Init static rx buffer num: 8
I (1055) wifi:Init dynamic rx buffer num: 32
I (1059) wifi_init: rx ba win: 6
I (1063) wifi_init: tcpip mbox: 32
I (1067) wifi_init: udp mbox: 6
I (1071) wifi_init: tcp mbox: 6
I (1074) wifi_init: tcp tx win: 5744
I (1079) wifi_init: tcp rx win: 5744
I (1083) wifi_init: tcp mss: 1440
I (1087) wifi_init: WiFi IRAM OP enabled
I (1091) wifi_init: WiFi RX IRAM OP enabled
I (1114) chip[DL]: NVS set: chip-counters/reboot-count = 1 (0x1)
I (1125) chip[DL]: NVS set: chip-counters/total-hours = 0 (0x0)
I (1128) chip[DL]: NVS set: chip-config/unique-id = "29CC4B94C13D6C08"
I (1129) chip[DL]: Real time clock set to 946684800 (0100/00/01 00:00:00 UTC)
D (1133) chip[DL]: CHIP event task running
I (1137) chip[DL]: Starting ESP WiFi layer
I (1141) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1246) wifi:mode : sta (78:21:84:93:f5:d4)
I (1247) wifi:enable tsf
W (1248) wifi:Haven't to connect to a suitable AP now!
I (1248) chip[DL]: Attempting to connect WiFi station interface
I (1254) chip[DL]: WiFi station state change: NotConnected -> Connecting
I (1259) chip[DL]: Done driving station state, nothing else to do...
I (1261) wifi:new:<8,0>, old:<1,0>, ap:<255,255>, sta:<8,0>, prof:9
I (1265) chip[SVR]: Initializing subscription resumption storage...
I (1272) wifi:state: init -> auth (b0)
I (1280) chip[SVR]: Server initializing...
D (1284) chip[FP]: Initializing FabricTable from persistent storage
I (1291) chip[TS]: Last Known Good Time: [unknown]
I (1295) chip[TS]: Setting Last Known Good Time to firmware build time 2023-05-14T19:31:49
I (1299) wifi:state: auth -> assoc (0)
I (1306) chip[DMG]: AccessControl: initializing
I (1311) chip[DMG]: Examples::AccessControlDelegate::Init
I (1316) chip[DMG]: AccessControl: setting
I (1320) chip[DMG]: DefaultAclStorage: initializing
I (1324) chip[DMG]: DefaultAclStorage: 0 entries loaded
D (1329) chip[IN]: UDP::Init bind&listen port=5540
D (1334) chip[IN]: UDP::Init bound to port=5540
D (1338) chip[IN]: UDP::Init bind&listen port=5540
D (1342) chip[IN]: UDP::Init bound to port=5540
D (1347) chip[IN]: BLEBase::Init - setting/overriding transport
I (1352) wifi:D (1352) chip[IN]: TransportMgr initialized
state: assoc -> run (10)
I (1367) chip[ZCL]: Using ZAP configuration...
I (1368) chip[DMG]: AccessControlCluster: initializing
I (1369) chip[ZCL]: 0x3f406b01ep 0 clus 0x0000_0030 attr 0x0000_0000 not supported
D (1376) chip[ZCL]: GeneralDiagnostics: OnDeviceReboot
I (1381) chip[ZCL]: Initiating Admin Commissioning cluster.
E (1386) chip[ZCL]: Duplicate attribute override registration failed
D (1394) chip[IN]: SecureSession[0x3ffc8a98]: Allocated Type:1 LSID:64648
D (1399) chip[SC]: Assigned local session key ID 64648
D (1404) chip[SC]: Waiting for PBKDF param request
I (1408) chip[SVR]: BLE networking available but BLE advertising is not supported
[... 144 lines cut ...]

E (6723) task_wdt: Print CPU 1 backtrace

Backtrace: 0x40082C6D:0x3FFBFE80 0x4008384D:0x3FFBFEA0 0x4000BFED:0x3FFD6090 0x40094419:0x3FFD60A0 0x40092DB1:0x3FFD60C0 0x40113508:0x3FFD6100 0x40113C25:0x3FFD6130 0x40113CD1:0x3FFD6150 0x401143B2:0x3FFD6170 0x400952B9:0x3FFD6190

  #0  0x40082C6D:0x3FFBFE80 in esp_crosscore_isr at C:\Users\fda\.platformio\packages\framework-espidf\components\esp_system/crosscore_int.c:92
  #1  0x4008384D:0x3FFBFEA0 in _xt_lowint1 at C:\Users\fda\.platformio\packages\framework-espidf\components\freertos\port\xtensa/xtensa_vectors.S:1114
  #2  0x4000BFED:0x3FFD6090 in ?? ??:0
  #3  0x40094419:0x3FFD60A0 in vPortClearInterruptMaskFromISR at C:\Users\fda\.platformio\packages\framework-espidf\components\freertos\port\xtensa\include/freertos/portmacro.h:571
      (inlined by) xTaskGetSchedulerState at C:\Users\fda\.platformio\packages\framework-espidf\components\freertos/tasks.c:4699
  #4  0x40092DB1:0x3FFD60C0 in xQueueSemaphoreTake at C:\Users\fda\.platformio\packages\framework-espidf\components\freertos/queue.c:1554
  #5  0x40113508:0x3FFD6100 in uartAvailable at C:/Users/fda/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:317 (discriminator 1)
  #6  0x40113C25:0x3FFD6130 in HardwareSerial::available() at C:/Users/fda/.platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.cpp:476
  #7  0x40113CD1:0x3FFD6150 in serialEventRun() at C:/Users/fda/.platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.cpp:116
  #8  0x401143B2:0x3FFD6170 in loopTask(void*) at C:/Users/fda/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:51 (discriminator 1)
  #9  0x400952B9:0x3FFD6190 in vPortTaskWrapper at C:\Users\fda\.platformio\packages\framework-espidf\components\freertos\port\xtensa/port.c:142

E (11723) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (11723) task_wdt:  - IDLE (CPU 1)
E (11723) task_wdt: Tasks currently running:
E (11723) task_wdt: CPU 0: IDLE
E (11723) task_wdt: CPU 1: loopTask
E (11723) task_wdt: Print CPU 0 (current core) backtrace
```
If I globally disable the watchdogs using `esp_task_wdt_init(-1, false)` the device keeps running and I'm able to commission the light successfully and control it. Of course I cannot do this in production. I'm using PlatformIO and believe that I have fulfilled all requirements mentioned in the readme.md

meta-space, 2023-05-30:
I think I found the culprit. 

I am using a custom espressif32-arduino build. It turns out that I had the option CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1=y in my sdkconfig. This will result in the watchdog being tripped when running Matter stuff. The [default esp32-arduino sdkconfig](https://github.com/espressif/arduino-esp32/blob/3670e2bf2aca822f2e1225fdb0e0796e490005a8/tools/sdk/esp32/sdkconfig#L961) doesn't enable this setting. 

I feel a little uneasy disabling the watchdog on the CPU1 Idle task... For the moment I'm going to assume that it is fine, as it is not enabled in the esp32-arduino-framework.

On the other hand the [Arduino startup](https://github.com/espressif/arduino-esp32/blob/3670e2bf2aca822f2e1225fdb0e0796e490005a8/cores/esp32/main.cpp#LL40C1-L53C2) shows that the loopTask first invokes setup() and then loop(). If loop() blocks for some reason I'd really like to know about it :) In the Light.ino it seems that the setup() function is blocking if the device hasn't been commissioned. 

```
void loopTask(void *pvParameters)
{
    setup();
    for(;;) {
#if CONFIG_FREERTOS_UNICORE
        yieldIfNecessary();
#endif
        if(loopTaskWDTEnabled){
            esp_task_wdt_reset();
        }
        loop();
        if (serialEventRun) serialEventRun();
    }
}
```

I also discovered interesting functions in [esp32-hal-misc.c](cores/esp32/esp32-hal-misc.c) which enable and disable the watchdogs at runtime. Maybe that's a better solution

Yacubane, 2023-07-24:
Hey! Thanks for sharing interesting info. In my private home project I also wanted more flexibility and created custom headers for handling watchdogs:

`watchdogs.h`
```
#pragma once
#include <esp_task_wdt.h>

class WatchdogsClass
{
public:
    WatchdogsClass() {}
    void setupWatchdog(uint32_t timeout) {
        esp_task_wdt_init(timeout, true);
        esp_task_wdt_add(NULL);
    }

    void restartWatchdog() {
        esp_task_wdt_reset();
    }
private:
};

extern WatchdogsClass Watchdogs;
```

`watchdogs.cpp`
```
#include "watchdogs.h"

WatchdogsClass Watchdogs;
```

Then you can use it inside your code
```
Watchdogs.setupWatchdog(30); // setup watchdogs for 30 seconds
Watchdogs.restartWatchdog(); // yield (restart watchdog counter)
```

closed 2023-07-24 by Yacubane