Yacubane/esp32-arduino-matter #34: Not a Matter-certified device

sshot999, 2023-06-21:
Anyone having a problem with Google failing to setup the device with the error "Not a Matter-certified device"

I have a fresh install of the library as of today and am using the latest ESP32 board as required in the readme.

Yacubane, 2023-07-24:
Yeah, it's true, you unfortunately have to make such "test" integration

closed 2023-07-24 by Yacubane

alpha815, 2023-08-03:
Hi, I am having same issue, uploaded the example code to esp32 and from Google Home i can discover and attempt to connect to it. At the end Google Home gave me same error. Can you please describe more about setting up test device in developer console?
From developer console i have added a new device with "test" vendor ID and unique Product ID. but can not find to relate it with esp32. do i have to add these vendor ID and Product ID to the library itself and recompile it?
Thanks.

Yacubane, 2023-08-06:
@alpha815 
the default data is:
[image]

TrevorSchirmer, 2023-09-17:
@Yacubane ,

How do you change the default VID/PID in your library? I know how to create the device in the Google Home Developer Console, just not sure how to apply it to the Arduino project.

Thanks!

Yacubane, 2023-09-17:
@TrevorSchirmer 
This library is just a wrapper of Matter library. You can change it for example by creating custom provider, however I haven't tried it and I can't send code -> I will send link to documentation:
https://github.com/espressif/esp-matter/blob/4707b88ceb5e6328f7b571b652487507429c1891/docs/en/developing.rst (chapter 2.5 Factory Data Providers section)

TrevorSchirmer, 2023-09-17:
@Yacubane Ok.

Thank you for setting this up, it is the easiest into to Matter I found by far.

-Trevor