Yacubane/esp32-arduino-matter #36: Opening & Closing commissioning window

smeisner, 2023-08-05:
I have been looking into a problem for days, looking thru the CHIP code and the ESP-matter code and cannot come up with a solution.

Problem: I am working on a device with a touch screen, so I can show a QR code for commissioning. This device (thermostat) also needs to;
a) set hostname
b) provide a web port
c) be able to change ssid & psk
d) retrieve wifi stats (ip addr, rssi, hostname, (dis)connected status)

With the above requirements, I cannot use the Arduino Wifi library AND the Matter code at the same time as they both try to initialize and start wifi. So I asked over on the esp-matter issues forum (https://github.com/espressif/esp-matter/issues/551) and the suggestion was to start Matter, close the commissioning window, cal Arduino Wifi.begin() and then reopen the comm window.

So, how do I open/close the commissioning window using esp-arduino-matter??

...or; How can I use esp-arduino-matter and provide the functionality in the list above?

smeisner, 2023-08-05:
Forgot to mention: I did try adding the following code:
```
void MatterCloseCommissioningWindow()
{
  ESP_LOGE(TAG, "Closing Commissioning Window\n");
  chip::CommissioningWindowManager & commissionMgr = chip::Server::GetInstance().GetCommissioningWindowManager();
  ESP_LOGE(TAG, "CommissionMgr = %p\n", commissionMgr);
  ESP_LOGE(TAG, "(Prior Close) Is open: %d\n", commissionMgr.IsCommissioningWindowOpen());
  commissionMgr.CloseCommissioningWindow();
  ESP_LOGE(TAG, "(After Close) Is open: %d\n", commissionMgr.IsCommissioningWindowOpen());
}
```
...and got linker errors:

```
/home/steve/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: lib/esp32-arduino-matter/src/esp32s3/libbt_nimble.a(bt.c.obj):(.literal.async_wakeup_request+0x4): undefined reference to `r_btdm_vnd_offload_post'
/home/steve/.platformio/packages/toolchain-xtensa-esp32s3@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: lib/esp32-arduino-matter/src/esp32s3/libbt_nimble.a(bt.c.obj): in function `async_wakeup_request':
/root/tools/esp-idf/components/bt/controller/esp32c3/bt.c:879: undefined reference to `r_btdm_vnd_offload_post
```

If I remove the added code, it compiles no problem.

smeisner, 2023-08-05:
BTW, the project I am working on is an open source ESP32-S3 based thermostat, including a custom PCB:
https://github.com/smeisner/smart-thermostat

The Matter code is in the fork named 'Matter'

It isn't ready for "prime time" quite yet, but I am close to having a reliable, functioning thermostat. The intelligence will come after implementing Matter properly. ...and maybe adding support for MQTT.

smeisner, 2023-08-06:
> I have been looking into a problem for days, looking thru the CHIP code and the ESP-matter code and cannot come up with a solution.
> 
> Problem: I am working on a device with a touch screen, so I can show a QR code for commissioning. This device (thermostat) also needs to; a) set hostname b) provide a web port c) be able to change ssid & psk d) retrieve wifi stats (ip addr, rssi, hostname, (dis)connected status)
> 
> With the above requirements, I cannot use the Arduino Wifi library AND the Matter code at the same time as they both try to initialize and start wifi. So I asked over on the esp-matter issues forum ([espressif/esp-matter#551](https://github.com/espressif/esp-matter/issues/551)) and the suggestion was to start Matter, close the commissioning window, cal Arduino Wifi.begin() and then reopen the comm window.
> 
> So, how do I open/close the commissioning window using esp-arduino-matter??
> 
> ...or; How can I use esp-arduino-matter and provide the functionality in the list above?

...or is there a way to retrieve the  `esp_netif_t` being used by Matter? With that, I can call the ESP-IDF wifi functions.

Yacubane, 2023-08-06:
Hey, really nice project!

This issue should be inside [esp-matter](https://github.com/espressif/esp-matter) or [connectedhomeip](https://github.com/project-chip/connectedhomeip) repo. The [esp32-arduino-matter](https://github.com/Yacubane/esp32-arduino-matter) is nothing more than pre-compiled set of these libraries.

Regarding your linker errors, it looks like pre-compiled NimBLE is missing some functions. Maybe they weren't compiled (and thus are not shipped with `esp32-arduino-matter` lib), because there was just no usage of them and compiler decided to not compile them at all. If you really want to do something production ready I would avoid this lib and use "raw" `esp-matter` with Arduino as component, especially if you want to modify default behavior of Matter device setup. This library is experimental and intended mostly for small, test devices.

smeisner, 2023-08-13:
@Yacubane Thanks for the replies! I am still making progress, except for one issue. Any time I try to call a Commisioning function, I get the linker error. I don't see how there was no usage of the NimBLE functions since I keep running into the problem.

I do realize esp32-arduino-matter is not fully ready for 'primetime', but I am trying to get a bunch of chunks of functionality working (like Rapid Prototyping idea) and esp32-arduino-matter is really helping a lot. I recently rewrote my entire wifi interface to get rid of the Arduino API in favor of the ESP-IDF interface. If possible I'd rather not do that yet with the Matter libraries.

I am not using BLE at all. Is there some way to get past these linker errors?

After rewriting the wifi module, I am able to control wifi (and get status) while Matter is running. Code is checked in to my 'Matter' branch in my repo.

emmby, 2023-08-14:
I can add a little bit of color to this.

I started having similar linking errors a couple of days ago, likely when I added a call to ConfigurationMgr.InitiateFactoryReset() (see https://github.com/espressif/esp-matter/issues/570).

For me at least, the problem is intermittent. I can link and upload just fine once or twice, but then I start getting linking errors.

To workaround, I just keep a copy of `esp32-arduino-matter.zip` handy and re-import it into Arduino IDE using Sketch > Include Library > Add .ZIP Library, overwriting whatever was there. This temporarily fixes the problem for the next couple of uploads, and then I do it again once it starts failing again.

I don't know why this is necessary, but it seems to work for a short time at least.

niiiksh, 2024-07-26:
> @Yacubane Thanks for the replies! I am still making progress, except for one issue. Any time I try to call a Commisioning function, I get the linker error. I don't see how there was no usage of the [...]

You mentioned rewriting wifi module. Do you have sources which uses wifi functionality with esp32-arduino-matter library?
I also see there is https://github.com/project-chip/connectedhomeip/pull/29017 fix to skip wifi stack init if already done. Does it mean I can use WiFi.begin(ssid,password) with Matter's CHIP state machine that connects to wifi?