Yacubane/esp32-arduino-matter #37: Disable chip logging?

emmby, 2023-08-21:
Since upgrading to 1.0.0-beta6, I've started getting a lot of chip logging polluting my serial port, eg.:

```
09:06:12.025 -> I (13992) chip[IM]: 
09:06:12.025 -> D (13994) chip[DMG]: 
09:06:12.025 -> D (13996) chip[DMG]: 
09:06:12.025 -> I (13998) chip[EM]: 
09:06:12.025 -> I (14000) chip[IN]: 
09:06:12.025 -> D (14002) chip[EM]: 
09:06:15.471 -> I (17456) chip[EM]: 
09:06:15.471 -> D (17456) chip[EM]: 
09:06:15.471 -> I (17457) chip[IN]:  
09:06:12.025 -> D (13996) chip[DMG]: 
09:06:12.025 -> I (13998) chip[EM]: 
09:06:12.025 -> I (14000) chip[IN]: 
09:06:12.025 -> D (14002) chip[EM]: 
09:06:15.471 -> I (17456) chip[EM]: 
09:06:15.471 -> D (17456) chip[EM]: 
09:06:15.471 -> I (17457) chip[IN]: 
09:06:15.471 -> I (17457) chip[EM]: 
09:06:15.471 -> I (17458) chip[IN]: 
09:06:15.471 -> D (17459) chip[EM]: 
09:06:15.503 -> I (17459) chip[SC]: 
09:06:15.503 -> D (17460) chip[SC]: 
09:06:15.503 -> E (17465) chip[SC]: CASE failed to match destination ID with local fabrics
09:06:15.503 -> E (17482) chip[IN]: CASE Session establishment failed: 10
09:06:15.503 -> D (17486) chip[IN]: 
09:06:15.503 -> D (17488) chip[IN]: 
09:06:15.503 -> D (17489) chip[SC]: 
09:06:15.534 -> E (17491) chip[EM]: OnMessageReceived failed, err = 10
09:06:15.534 -> I (17529) chip[EM]: 
09:06:15.534 -> D (17529) chip[EM]: 
09:06:15.534 -> D (17530) chip[EM]: 
09:06:17.859 -> E (19844) chip[IN]: Data received on an unknown session (LSID=35276). Dropping it!
```

Is it possible to turn these logs off or are they just built into the binary now? I tried using  `esp_log_level_set("*", ESP_LOG_ERROR)` but the logs keep coming

Yacubane, 2023-08-21:
I have enabled more logs when compiling this library here:
https://github.com/Yacubane/esp32-arduino-matter-builder/commit/3a3eb8ac5d767c5e1fa6e78ecf4f3aa590a6e3ef

Unfortunately by nature of this library (it's precompiled) you won't be able to change log level of library. You would have to compile it again.

Maybe it would be worth to provide library in two precompiled versions – one with all logs and one with restricted logs.

Keep in mind that this library is experimental, have a lot of limitations and should be used for testing purposes only :>

emmby, 2023-08-21:
I like the idea of a second version with debug and info logs turned off (both are pretty spammy). My project is experimental, but it does make my own debugging more difficult since the Arduino IDE is not very sophisticated and doesn't allow filtering/searching in the Serial Monitor.

Yacubane, 2023-11-26:
I will look at this when migrating to Arduino ESP32 core 3.0.0, when it releases, because this will probably mean more changes in sdkconfigs

Kitchenery-DLoranger, 2024-04-03:
This is a great project, making my life a lot easier being a HW engineer learning to do this firmware for matter.

Seems the new core has been released, any chance you can do this build with the chipmessages able to be muted?  

https://blog.espressif.com/announcing-the-arduino-esp32-core-version-3-0-0-3bf9f24e20d4

Yacubane, 2024-04-03:
> This is a great project, making my life a lot easier being a HW engineer learning to do this firmware for matter.
>
> Seems the new core has been released, any chance you can do this build with the [...]

According to https://github.com/espressif/arduino-esp32/releases it's still in alpha. I though that till now we will have stable version but there isn't. Maybe it would be better to stick to older version?

Dloranger, 2024-04-05:
Will create a new issue with the challenge I am facing so as not to pollute this thread.