Yacubane/esp32-arduino-matter #41: Arduino IDE update 2.2.1 breaks esp-arduino-matter

frios, 2023-09-13:
A recent update to the Arduino Ice (2.2.1) seems to have broken this library. Compile of any sample gives an extensive list of errors. e.g. A simple compile of the Light sample gives the following errors:

chip::BitMask<AlarmMap> state    = static_cast<chip::BitMask<AlarmMap>>(0);
                                                                  ^~~~~~~~
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:16691:66: error: template argument 2 is invalid
     chip::BitMask<AlarmMap> mask     = static_cast<chip::BitMask<AlarmMap>>(0);
                                                                  ^~~~~~~~
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:18181:30: error: template argument 2 is invalid
     chip::BitMask<DaysMaskMap> daysMask = static_cast<chip::BitMask<DaysMaskMap>>(0);
                              ^
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:18181:69: error: template argument 2 is invalid
     chip::BitMask<DaysMaskMap> daysMask = static_cast<chip::BitMask<DaysMaskMap>>(0);
                                                                     ^~~~~~~~~~~
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:18202:30: error: template argument 2 is invalid
     chip::BitMask<DaysMaskMap> daysMask = static_cast<chip::BitMask<DaysMaskMap>>(0);
                              ^
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:18202:69: error: template argument 2 is invalid
     chip::BitMask<DaysMaskMap> daysMask = static_cast<chip::BitMask<DaysMaskMap>>(0);
                                                                     ^~~~~~~~~~~
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:18268:28: error: template argument 2 is invalid
     Optional<chip::BitMask<DaysMaskMap>> daysMask;
                            ^~~~~~~~~~~
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:18268:39: error: template argument 1 is invalid
     Optional<chip::BitMask<DaysMaskMap>> daysMask;
                                       ^~
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:18290:28: error: template argument 2 is invalid
     Optional<chip::BitMask<DaysMaskMap>> daysMask;
                            ^~~~~~~~~~~
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:18290:39: error: template argument 1 is invalid
     Optional<chip::BitMask<DaysMaskMap>> daysMask;
                                       ^~
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:19246:95: error: template argument 2 is invalid
     using Type             = chip::BitMask<chip::app::Clusters::DoorLock::DlCredentialRuleMask>;
                                                                                               ^
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:19247:95: error: template argument 2 is invalid
     using DecodableType    = chip::BitMask<chip::app::Clusters::DoorLock::DlCredentialRuleMask>;
                                                                                               ^
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:19248:95: error: template argument 2 is invalid
     using DecodableArgType = chip::BitMask<chip::app::Clusters::DoorLock::DlCredentialRuleMask>;
                                                                                               ^
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:19331:100: error: template argument 2 is invalid
     using Type             = chip::BitMask<chip::app::Clusters::DoorLock::DlSupportedOperatingModes>;
                                                                                                    ^
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:19332:100: error: template argument 2 is invalid
     using DecodableType    = chip::BitMask<chip::app::Clusters::DoorLock::DlSupportedOperatingModes>;
                                                                                                    ^
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:19333:100: error: template argument 2 is invalid
     using DecodableArgType = chip::BitMask<chip::app::Clusters::DoorLock::DlSupportedOperatingModes>;
                                                                                                    ^
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:19343:105: error: template argument 2 is invalid
     using Type             = chip::BitMask<chip::app::Clusters::DoorLock::DlDefaultConfigurationRegister>;
                                                                                                         ^
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:19344:105: error: template argument 2 is invalid
     using DecodableType    = chip::BitMask<chip::app::Clusters::DoorLock::DlDefaultConfigurationRegister>;
                                                                                                         ^
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:19345:105: error: template argument 2 is invalid
     using DecodableArgType = chip::BitMask<chip::app::Clusters::DoorLock::DlDefaultConfigurationRegister>;
                                                                                                         ^
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:19403:101: error: template argument 2 is invalid
     using Type             = chip::BitMask<chip::app::Clusters::DoorLock::DlLocalProgrammingFeatures>;
                                                                                                     ^
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:19404:101: error: template argument 2 is invalid
     using DecodableType    = chip::BitMask<chip::app::Clusters::DoorLock::DlLocalProgrammingFeatures>;
                                                                                                     ^
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:19405:101: error: template argument 2 is invalid
     using DecodableArgType = chip::BitMask<chip::app::Clusters::DoorLock::DlLocalProgrammingFeatures>;
                                                                                                     ^
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:19537:55: error: 'DecodableType' in 'struct chip::app::Clusters::DoorLock::Attributes::CredentialRulesSupport::TypeInfo' does not name a type
         Attributes::CredentialRulesSupport::TypeInfo::DecodableType credentialRulesSupport =
                                                       ^~~~~~~~~~~~~
[... 154 lines cut ...]
     using DecodableType    = chip::BitMask<chip::app::Clusters::FanControl::RockBitmap>;
                                                                                       ^
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:22596:87: error: template argument 2 is invalid
     using DecodableArgType = chip::BitMask<chip::app::Clusters::FanControl::RockBitmap>;
                                                                                       ^
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:22606:87: error: template argument 2 is invalid
     using Type             = chip::BitMask<chip::app::Clusters::FanControl::WindBitmap>;
[... cut ...]
                                                                                       ^
/Users/feldy/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/app-common/zap-generated/cluster-objects.h:22619:87: error: template argument 2 is invalid
     using DecodableType    = chip::BitMask<chip::app::Clusters::FanControl::WindBitmap>;
                                                                                       ^
... text removed to fit body limitations

exit status 1

Compilation error: exit status 1

Yacubane, 2023-09-13:
Thank you, I will look into this in my spare time

Yacubane, 2023-09-17:
Hmm... I have just updated Arduino IDE to 2.2.1 and it works

Can you send full config? Which board is selected in Arduino IDE? Did you enable C++17 for Arduino IDE? What version of ESP32 boards are you using?

frios, 2023-09-17:
I got it working. It seems like I reinstalled the IDE and that set the C++ back to 11. Once I followed your directions again, it all seems to have fallen back into place. Thank you.

closed 2023-09-18 by frios