Yacubane/esp32-arduino-matter #42: Default examples not working

zhekch, 2023-09-16:
When attempting to run the default examples, they all fail to compile. I get this error with the light example.
(this is only the last part of the error, as it is too big so i didnt paste it here)

```
/Users/zhekch/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/esp_matter_attribute_utils.h: In instantiation of 'nullable<T>::nullable(T) [with T = unsigned char]':
/Users/zhekch/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/esp_matter_feature.h:100:88:   required from here
/Users/zhekch/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/esp_matter_attribute_utils.h:184:62: error: 'IsNullValue' is not a member of 'chip::app::NumericAttributeTraits<unsigned char, false>'
/Users/zhekch/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/esp_matter_attribute_utils.h: In instantiation of 'nullable<T>::nullable(T) [with T = short int]':
/Users/zhekch/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/esp_matter_cluster.h:385:106:   required from here
/Users/zhekch/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/esp_matter_attribute_utils.h:184:62: error: 'IsNullValue' is not a member of 'chip::app::NumericAttributeTraits<short int, false>'
/Users/zhekch/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/esp_matter_attribute_utils.h: In instantiation of 'void nullable<T>::operator=(T) [with T = unsigned char]':
/Users/zhekch/Documents/Arduino/libraries/ESP32_Arduino_Matter/examples/Light/Light.ino:73:50:   required from here
/Users/zhekch/Documents/Arduino/libraries/ESP32_Arduino_Matter/src/esp_matter_attribute_utils.h:218:62: error: 'IsNullValue' is not a member of 'chip::app::NumericAttributeTraits<unsigned char, false>'
         if (chip::app::NumericAttributeTraits<T>::IsNullValue(value)) {
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~

exit status 1

Compilation error: exit status 1
```

Yacubane, 2023-09-17:
Can you send full config? Which board is selected in Arduino IDE? Did you enable C++17 for Arduino IDE?

zhekch, 2023-09-24:
Here is my board config:
[image]
I believe enabling c++ is a windows feature, while im on mac, and what exactly do you mean by a version of a board?

WillemJoosten, 2023-09-28:
Same issues here, with the same error on the light example (MacOS) as @zhekch 

running here:
MacOS 13.5.2 Ventura
Arduino IDE 2.2.1
C++17 enabled 
Board: ESP32 Dev Module
Espressif ESP32 boards package: 2.0.11

WillemJoosten, 2023-09-29:
Managed to get it to compile:
you MUST use the C++17 modification as mentioned in the readme.
I also struggled to get it to compile on my Mac (Ventura 13.6) but C++ is not a windows-only feature :)

Yacubane, 2023-11-26:
@zhekch did you manage to solve this issue? As @WillemJoosten  said, C++ is not a windows-only feature 😊

Martenz, 2024-07-22:
Got this error on my Mac Book Pro with platformio: **'OnOff' has not been declared**

I added to the platform.ini as described in the readme:

`build_unflags=-std=gnu++11
build_flags = `

It won't recognize these classes

- OnOff
- attribute_t

plus other names.

Thanks