Yacubane/esp32-arduino-matter #53: Linking problem in platformio

teomona98, 2024-03-14:
Hi, I'm having a strange problem and i don't know what to try.
It seems like there's some symbols being redeclared somewhere, but i can't seem to get a hold of the problem.

If i include the Matter.h header file from main.cpp it works perfectly fine but as soon as i include in from any other header file in the project i get the following errors at linking time:
`/Users/matteomoneta/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/az-delivery-devkit-v4/src/me_core/me_component/me_component_on_off.cpp.o: in function `emberAfLevelControlClusterMoveToClosestFrequencyCallback(chip::app::CommandHandler*, chip::app::ConcreteCommandPath const&, chip::app::Clusters::LevelControl::Commands::MoveToClosestFrequency::DecodableType const&)':
/Users/matteomoneta/Documents/ME/me01-esp-fw/.pio/libdeps/az-delivery-devkit-v4/ESP32 Arduino Matter/src/esp32-arduino-matter_patches_matter.h:12: multiple definition of `emberAfLevelControlClusterMoveToClosestFrequencyCallback(chip::app::CommandHandler*, chip::app::ConcreteCommandPath const&, chip::app::Clusters::LevelControl::Commands::MoveToClosestFrequency::DecodableType const&)'; .pio/build/az-delivery-devkit-v4/src/me_core/me_component/me_component.cpp.o:/Users/matteomoneta/Documents/ME/me01-esp-fw/.pio/libdeps/az-delivery-devkit-v4/ESP32 Arduino Matter/src/esp32-arduino-matter_patches_matter.h:12: first defined here
/Users/matteomoneta/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/az-delivery-devkit-v4/src/me_core/me_component/me_component_on_off.cpp.o:(.bss._ZN12MatterUnused1_E+0x0): multiple definition of `MatterUnused::_'; .pio/build/az-delivery-devkit-v4/src/me_core/me_component/me_component.cpp.o:(.bss._ZN12MatterUnused1_E+0x0): first defined here
/Users/matteomoneta/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/az-delivery-devkit-v4/src/me_core/me_controller/me_controller.cpp.o: in function `emberAfLevelControlClusterMoveToClosestFrequencyCallback(chip::app::CommandHandler*, chip::app::ConcreteCommandPath const&, chip::app::Clusters::LevelControl::Commands::MoveToClosestFrequency::DecodableType const&)':
/Users/matteomoneta/Documents/ME/me01-esp-fw/.pio/libdeps/az-delivery-devkit-v4/ESP32 Arduino Matter/src/esp32-arduino-matter_patches_matter.h:12: multiple definition of `emberAfLevelControlClusterMoveToClosestFrequencyCallback(chip::app::CommandHandler*, chip::app::ConcreteCommandPath const&, chip::app::Clusters::LevelControl::Commands::MoveToClosestFrequency::DecodableType const&)'; .pio/build/az-delivery-devkit-v4/src/me_core/me_component/me_component.cpp.o:/Users/matteomoneta/Documents/ME/me01-esp-fw/.pio/libdeps/az-delivery-devkit-v4/ESP32 Arduino Matter/src/esp32-arduino-matter_patches_matter.h:12: first defined here
/Users/matteomoneta/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/az-delivery-devkit-v4/src/me_core/me_controller/me_controller.cpp.o:(.bss._ZN12MatterUnused1_E+0x0): multiple definition of `MatterUnused::_'; .pio/build/az-delivery-devkit-v4/src/me_core/me_component/me_component.cpp.o:(.bss._ZN12MatterUnused1_E+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/az-delivery-devkit-v4/firmware.elf] Error 1`

I've tried to exclude it from every file and as soon as I do that, it links perfectly fine.
Thank's in advance.

DaniilVdovin, 2024-09-06:
Изменил конфигурацию в `platform.ini`
`platform = espressif32@6.6.0 `

это помогло мне решить проблему!

VinicioAM, 2024-10-07:
> Изменил конфигурацию в `platform.ini` `platform = espressif32@6.6.0 `
> 
> это помогло мне решить проблему!

Did not work to me.
I Have the same problem here.

mohammedbinsaleem00, 2025-01-31:
I've been having the same issue. anyone figure out how to fix it?