Yacubane/esp32-arduino-matter #58: Question to Versions

RobAllIsGood, 2024-04-18:
Hi,

I added my Espresso machine to my Google  Home using Matter and ESP32 C3 thanks to your work here :).
Wrote a little table driven state machine to go through the steps in the correct order and with the correct timing.

I know you properly have other thinsgs on your plate.
Curious about if development will continue?

Saw that Release 3.0.0RC1 is out. (link https://github.com/espressif/arduino-esp32/releases ).
Also 2.0.15 has been out, tried that, doesnt seem to work.

Cheers

Rob :)

Yacubane, 2024-04-18:
Hey, yes development will continue, I was waiting for 3.0.0 to release but it took longer than I expected :)

Thank you for information, it is now worth to start developing support for 3.0.0.

RobAllIsGood, 2024-04-18:
Awesome, love your work, happy to contribute my little code to your example library if you like. Write it generically so other automation tasks can be accomplished where a number of outputs need to be swutched in timed sequence.

hassbian-ABC, 2024-04-20:
> 嘿，是的，开发会继续，我一直在等待3.0.0的发布，但花的时间比我预期的要长:)
> 
> 感谢您提供的信息，现在值得开始开发3.0.0的支持。

👍🏻