Yacubane/esp32-arduino-matter #59: Issue on default execution of "Light" -- wifi: invalid magic number 7 error

arcangeI7, 2024-05-07:
Hello, I am new to the world of Matter and ESP32. I will try to be brief and give all the important details. If I left something out, don't hesitate to tell me.
I installed the library and loaded the Light example. Once it has been compiled and written on the ESP32, it started to run, but the code crashes with some error with the Wi-Fi connection (or at least that's what I understand).
I attach a screenshot of the error.
[image]

I'm using the Arduino IDE with version 1.8.19 (I know it's old, but frankly I like it much more than the current versions)
I would appreciate it if someone could help me see if I'm just stupid and overlooking something obvious or if it's a bigger problem.

My apologies if I have bad spelling, my English is self-taught :)

niiiksh, 2024-06-21:
I have the same problem too.
What version of esp32 boards package do you use? I'm on 2.0.17 but I guess I should try to match version of the Matter library that was built with 2.0.14

niiiksh, 2024-06-21:
Yes, with version 2.0.14 this error disappeared and it ran successfully